contiki-os/contiki #939: there is no obvious version file in source tree

mcr, 2015-01-05:
I suggest that version.h or VERSION.md or some such should exist prominently in the top-level.
For releases the contents should be obvious.  For non-releases (i.e. git-master), it could contain something that says, something like "PRE-3.0-GIT", perhaps include a year/month.  Ideally, the build system would include the git id for that.

gebart, 2015-01-08:
A macro is defined by the `Makefile` which is passed via `CFLAGS` to all compilation modules, e.g. `-DCONTIKI_VERSION_STRING=\"Contiki-2.6-1877-gce49276\"`

g-oikonomou, 2015-01-11:
There's also `core/contiki-version.h` which provides a fallback if the build system won't define `CONTIKI_VERSION_STRING` (which I believe is the case for cygwin builds)

dak664, 2015-01-11:
The git version define is picket up by my cygwin. If it does not work remove the redirect, See $CONTIKI/Makefile.include:

 ### Check for a git repo and pass version if found
 ### git.exe in Windows cmd shells may require no stderr redirection
 #RELSTR=${shell git describe --tags}
  RELSTR=${shell git describe --tags 2>/dev/null}
 ifneq ($(RELSTR),)
 CFLAGS += -DCONTIKI_VERSION_STRING=\"Contiki-$(RELSTR)\"
 endif

atiselsts, 2015-01-12:
Having the version as a string isn't really that helpful.

As Contiki API is changing all the time, code like this is required:

    #if CONTIKI_MAJOR_VERSION_NUMBER >= 3
    // Version 3.0+
    #include "net/linkaddr.h"
    #else
    // Version 2.x
    #include "net/rime/rimeaddr.h"
    #endif

Since it is not possible to make a preprocessor conditional on a string, the user has to define `CONTIKI_MAJOR_VERSION_NUMBER` on his own, which is a PITA.

@dak664 `shell git describe --tags` is a good idea. What is strange is that it returns a string starting with `2.6` for the git version and `2.7` for the release version.

cmorty, 2015-01-15:
@dak664 The code changed to
```sh
### Check for a git repo and pass version if found
### git.exe in Windows cmd shells may require no stderr redirection
ifndef RELSTR
RELSTR:=${shell git --git-dir ${CONTIKI}/.git describe --tags --always}
endif
```
Are you sure you have a current version of cooja?

gebart, 2015-01-16:
The reason for the 2.6-xxxxxx when using current master is that the commit that was tagged 2.7 is not part of the master branch, it was branched off from master, and no new tags have been created since then, so when git is looking for the latest tag going backwards from current master it will only find the 2.6 tag.

This is the commit:
https://github.com/contiki-os/contiki/commit/763590695ff6ee5a281895f125984dc4bf38efdf

simonduq, 2015-08-21:
It seems we're still having this problem, i.e. the latest common tag always points to 2.6. 
I would vote for either simply removing `-DCONTIKI_VERSION_STRING` from `makefile.include` (will fall back to the definition in `core/contiki-version.h`)

alignan, 2015-08-21:
I'm annoyed by the 2.6-xxxx references also, but I agree with @atiselsts about version as strings not being too helpful either, but for the moment I'm OK with @simonduq

laurentderu, 2015-08-24:
I'm a bit reluctant to remove CONTIKI_VERSION_STRING, it's really helpful to know the exact version of the firmware you're running, "Contiki-3.0" does tell you much on the other hand.

Could we merge the 3.0 branch in master and tag it there instead ?

laurentderu, 2015-08-24:
Forget the merge, I see that new code has already been merged into master.

simonduq, 2015-08-25:
How about still relying on `RELSTR` but adding a tag "3.x" at the commit where we switched from 2.x to 3.x?
This way we get "3.0" when using the release, "3.x-count-hash" otherwise

laurentderu, 2015-08-25:
Good compromise I think. Otherwise we could use the version from contiki-version and append the current commit (using something like git rev-parse --short=8 HEAD) but you have less info.

alignan, 2015-08-25:
:+1: @laurentderu

simonduq, 2015-08-26:
so shall we add a tag "3.x" on this commit: https://github.com/contiki-os/contiki/commit/ae5a3f9f677e64b9a5b8596db4926f5e53ff404f ?

cmorty, 2015-09-02:
@simonduq ->  #1232 -> :+1:  At least as a quick fix.

adamdunkels, 2015-09-02:
@simonduq: I'll try making a reference tag `3.x` on that commit.

adamdunkels, 2015-09-02:
I tagged and pushed, I now see this printout from the `native` target:

```
$ ./hello-world.native 
Contiki-3.x-1416-gc792993 started with IPV6, RPL
```

So seems to work!

closed 2015-09-02 by adamdunkels