contiki-os/contiki #941: Issue with nullsec_driver

charly0, 2015-01-13:
Hello,

I have some issues when i want to compile any examples (rpl-border-router-, hello-world...) on the lastest version of contiki on my avr-atmega128rfa1.

Since I defined UIP_CONF_IPV6 to 1, I decided to set the NETSTACK_NETWORK variable on sicslowpan_driver in the file core/net/netstack.h 

The result of my command is :

root@PC1:/home/pc1/git/contiki-aes/examples/ipv6/rpl-border-router# make all TARGET=avr-atmega128rfa1 WITH_SLIP=1 WITH_WEBSERVER=0
  AR        contiki-avr-atmega128rfa1.a
  CC        border-router.c
  LD        border-router.avr-atmega128rfa1
contiki-avr-atmega128rfa1.a(sicslowpan.o): In function `send_packet':
/home/pc1/git/contiki-aes/examples/ipv6/rpl-border-router/../../../core/net/ipv6/sicslowpan.c:1346: undefined reference to `nullsec_driver'
/home/pc1/git/contiki-aes/examples/ipv6/rpl-border-router/../../../core/net/ipv6/sicslowpan.c:1346: undefined reference to `nullsec_driver'
contiki-avr-atmega128rfa1.a(sicslowpan.o): In function `output':
/home/pc1/git/contiki-aes/examples/ipv6/rpl-border-router/../../../core/net/ipv6/sicslowpan.c:1449: undefined reference to `nullsec_driver'
/home/pc1/git/contiki-aes/examples/ipv6/rpl-border-router/../../../core/net/ipv6/sicslowpan.c:1449: undefined reference to `nullsec_driver'
contiki-avr-atmega128rfa1.a(nullmac.o): In function `packet_input':
/home/pc1/git/contiki-aes/examples/ipv6/rpl-border-router/../../../core/net/mac/nullmac.c:57: undefined reference to `nullsec_driver'
contiki-avr-atmega128rfa1.a(nullmac.o):/home/pc1/git/contiki-aes/examples/ipv6/rpl-border-router/../../../core/net/mac/nullmac.c:57: more undefined references to `nullsec_driver' follow
collect2: error: ld returned 1 exit status
make: *** [border-router.avr-atmega128rfa1] Erreur 1
rm border-router.co

Thank you for your help

darkopetrovic, 2015-01-16:
The `nullsec_driver` is found in `core/net/llsec/nullsec.c `. 

Its seems that the module is missing for this platform, try adding `core/net/llsec` in the makefile Makefile.avr-atmega128rfa1:  

`MODULES += core/net/mac core/net core/net/mac/sicslowmac core/net/mac/contikimac core/net/llsec`

AkhilaCA, 2015-04-24:
Hi i am working on contiki-2.7 with avr-atmega128rfa12. I compiled the rpl-border-router code for avr-atmega128rfa1 platform sucessfully with thes changes in the make file

1. MODULES += core/net/mac core/net core/net/mac/sicslowmac core/net/mac/contikimac core/net/llsec
Add this line to the Makefile of platform/avr-atmega128rfa1/Makefile.avr-atmega128rfa1

2. CFLAGS+= -DUIP_CONF_IPV6_RPL 
Add this flag to Makefile of examples/ipv6/rpl-border-router/Makefile

I want to do 6lowpan application in avr-atmega128rfa1 using coniki. For that I use tunslip6.c (tools/tunslip6)

While running tunslip6 : sudo ./tunslip6 -B 38400 -s /dev/ttyUSB0 aaaa::1/64

The output is :

********SLIP started on ``/dev/ttyUSB0''
opened tun device ``/dev/tun0''
ifconfig tun0 inet `hostname` up
ifconfig tun0 add aaaa::1/64
ifconfig tun0 add fe80::0:0:0:1/64
ifconfig tun0

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:127.0.1.1  P-t-P:127.0.1.1  Mask:255.255.255.255
          inet6 addr: fe80::1/64 Scope:Link
          inet6 addr: aaaa::1/64 Scope:Global
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

*** Address:aaaa::1 => aaaa:0000:0000:0000
Got configuration message of type P
Setting prefix aaaa::
hai hereServer IPv6 addresses:
 aaaa::ff:fe00:1
 fe80::ff:fe00:1

Then I uploaded the udp-client (rpl-border-router/rpl-udp/udp-client.c)

When I tried to discover the udp-client it shows the same ip of the border router.
aaaa::ff:fe00:1

What wrong with this. Am I doing in correct way.Help me
Thank you in advance.

AkhilaCA, 2015-04-24:
Hi i am working on contiki-2.7 with avr-atmega128rfa12. I compiled the rpl-border-router code for avr-atmega128rfa1 platform sucessfully with thes changes in the make file

MODULES += core/net/mac core/net core/net/mac/sicslowmac core/net/mac/contikimac core/net/llsec
Add this line to the Makefile of platform/avr-atmega128rfa1/Makefile.avr-atmega128rfa1

CFLAGS+= -DUIP_CONF_IPV6_RPL 
Add this flag to Makefile of examples/ipv6/rpl-border-router/Makefile

I want to do 6lowpan application in avr-atmega128rfa1 using coniki. For that I use tunslip6.c (tools/tunslip6)

While running tunslip6 : sudo ./tunslip6 -B 38400 -s /dev/ttyUSB0 aaaa::1/64

The output is :

********SLIP started on /dev/ttyUSB0'' opened tun device/dev/tun0''
ifconfig tun0 inet hostname up
ifconfig tun0 add aaaa::1/64
ifconfig tun0 add fe80::0:0:0:1/64
ifconfig tun0

tun0 Link encap:UNSPEC HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00

inet addr:127.0.1.1 P-t-P:127.0.1.1 Mask:255.255.255.255
inet6 addr: fe80::1/64 Scope:Link
inet6 addr: aaaa::1/64 Scope:Global
UP POINTOPOINT RUNNING NOARP MULTICAST MTU:1500 Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:500 
RX bytes:0 (0.0 B) TX bytes:0 (0.0 B)

*** Address:aaaa::1 => aaaa:0000:0000:0000
Got configuration message of type P
Setting prefix aaaa::
hai hereServer IPv6 addresses:
aaaa::ff:fe00:1
fe80::ff:fe00:1

Then I uploaded the udp-client (rpl-border-router/rpl-udp/udp-client.c)

When I tried to discover the udp-client it shows the same ip of the border router.
aaaa::ff:fe00:1

What wrong with this. Am I doing in correct way.Help me
Thank you in advance.