contiki-os/contiki #942: problem in connecting cc2538dk to contiki

latapata, 2015-01-13:
hello All,

I want to communicate my cc2538dk with contiki so that i can upload the code directly by typing make %.upload

I followed all the steps but when I type 

sudo echo 0403 a6d1 > /sys/bus/usb-serial/drivers/ftdi_sio/new_id
or 
echo 0403 a6d1 > /sys/bus/usb-serial/drivers/ftdi_sio/new_id
it says 
bash: /sys/bus/usb-serial/drivers/ftdi_sio/new_id: Permission denied

latapata, 2015-01-13:
I found out that this file only has root permissions and the root doesn’t actually have a password set.
Type the following command:

sudo passwd

The following message will appear:

[sudo] password for [username]: [Type your user password and press return]

After that another message will appear, the following:

Type new UNIX password: [Type the root password you want]

Now the last thing you need to do, the following:

Retype new UNIX password: [Retype the root password you chosen before]

The following message will appear:

passwd: password updated successfully

If the above message showed up on your terminal, you can now use the root privileges by typing:

su

and then type 

echo 0403 a6d1 > /sys/bus/usb-serial/drivers/ftdi_sio/new_id

latapata, 2015-01-13:
I did everything but 
make cc2538-demo.upload does not work. 

says 

make: *** No rule to make target `cc2538-demo.upload'.  Stop.

nvt, 2015-01-14:
Support questions can be sent to the contiki-developers mailing list.

closed 2015-01-14 by nvt