contiki-os/contiki #948: RPL DAO Timer Syntax Error

tjrendall, 2015-01-26:
I have found that in:

rpl/rpl-timers.c,

in the function:

set_dao_lifetime_timer,

there is a multiplication that does not compile properly on my machine.

on the line:

> expiration_time = (clock_time_t)instance->default_lifetime *
      (clock_time_t)instance->lifetime_unit *
      CLOCK_SECOND / 2;

I have that expiration_time = 0, with non-zero default_lifetime and lifetime_unit.

However, when I add parentheses as so:

expiration_time = ((clock_time_t)instance->default_lifetime) *((clock_time_t)instance->lifetime_unit) *(CLOCK_SECOND / 2);

expiration_time returns the proper value.

I'd welcome anyone else to check this for themselves. The in-error mode causes continuous (0-interval timers) RPL ICMPv6 DAO messages to be sent, causing major power losses.

Jordan

gebart, 2015-01-26:
Which platform are you using?

tjrendall, 2015-01-26:
I'm running it on an AVR atmega256RFR2 and have made it possible to debug using Atmel Studio so I have a more visual development environment than just compiling with a Linux terminal.

gebart, 2015-01-27:
@tjrendall is that `TARGET=avr-atmega128rfa1` ?

tjrendall, 2015-01-27:
@gebart actually, no. I made the changes in the radio folder for avr to replace ATMEGA128RFA1 with ATMEGA256RFR2, wherever I saw it. I already had developed boards with this MCU before attempting to switch to Contiki OS.

nvt, 2015-02-26:
I suppose it is enough to ensure that the division by 2 is not the last step in the calculation; or, preferably, to cast the variables to a sufficiently large integer type instead of the platform-dependent clock_time_t.

The added parentheses around the lifetime variable reads shouldn't be necessary.