contiki-os/contiki #949: Contiki 3.x won't compile, PACKETBUF_*... undeclared in collect.h

Azeirah, 2015-01-28:
I'm trying to build the contiki 3.x branch pulled from master today for the avr-atmega128rfa1 target, however I'm getting a compiler error.

My makefile looks as follows: 

```
CONTIKI_PROJECT=domotica
$TARGET = avr-atmega128rfa1

UIP_CONF_IPV6_RPL=1

PROJECTDIRS=source
PROJECT_SOURCEFILES = SPI.c tcp_ip.c device-registration.c jsmn.c json-rpc.c server_calls.c util.c
all: $(CONTIKI_PROJECT)

MODULES += core/net/ipv6 core/net/ipv4 core/net/ip core/net/mac core/net core/net/rime core/net/rpl core/net/mac/sicslowmac core/net/contikimac core/net/llsec

CONTIKI_CONF_RANDOM_MAC=1

UIP_CONF_LL_802154=1

UIP_CONF_IPV6=1
CFLAGS+= -DUIP_CONF_IPV6_RPL=1
CFLAGS+= -DUIP_CONF_IPV6=1

CONTIKI=contiki
include $(CONTIKI)/Makefile.include
```

Contiki release 2.7 compiles just fine with the same makefile.

Before this compiler error, I had a different one mentioning I was missing the nullsec drivers, according to a post I found I should add some modules to the makefile, `MODULES += core/net/ipv6 core/net/ipv4 core/net/ip core/net/mac core/net core/net/rime core/net/rpl core/net/mac/sicslowmac core/net/contikimac`.

I've got 6 compiler errors here

```
Error	10	'PACKETBUF_ADDR_ESENDER' undeclared here (not in a function)	[shortened]\domotica\contiki\core\net\rime\collect.h	94	31	domotica
Error	12	'PACKETBUF_ATTR_EPACKET_ID' undeclared here (not in a function)	[shortened]\domotica\contiki\core\net\rime\collect.h	95	31	domotica
Error	32	'PACKETBUF_ATTR_ERELIABLE' undeclared (first use in this function)	[shortened]\domotica\contiki\core\net\rime\collect.c	892	22	domotica
Error	16	'PACKETBUF_ATTR_HOPS' undeclared here (not in a function)	[shortened]\domotica\contiki\core\net\rime\collect.h	98	31	domotica
Error	18	'PACKETBUF_ATTR_MAX_REXMIT' undeclared here (not in a function)	[shortened]\domotica\contiki\core\net\rime\collect.h	99	31	domotica
Error	14	'PACKETBUF_ATTR_TTL' undeclared here (not in a function)	[shortened]\domotica\contiki\core\net\rime\collect.h	97	31	domotica
```
Note: I took away the irrelevant part of the path.

vac, 2015-02-02:
I have solve similar issue by setting:
`#define NETSTACK_CONF_WITH_RIME 1`
in my platform "contiki-conf.h"
hope that it would help in your case (probably if you use rime_driver as rime_driver)

simonduq, 2015-02-02:
@Azeirah, is the makefile yours or is it from the Contiki tree?
It seems outdated anyway, one shouldn't set `UIP_CONF_IPV6` and `-DUIP_CONF_IPV6_RPL` anymore. Now the different netstacks are enabled with `NETSTACK_CONF_WITH_RIME `, `NETSTACK_CONF_WITH_IPV4`, `NETSTACK_CONF_WITH_IPV6`. 

In your case I guess what you need is:
`#define NETSTACK_CONF_WITH_IPV6 1`

closed 2015-06-19 by g-oikonomou

mhshd, 2019-05-17:
@vac I have the same problem. I added #define NETSTACK_CONF_WITH_RIME 1 to the contiki-conf.h file in exp5438 platform folder, but I still get the error. here is the makefile :
` CONTIKI_PROJECT = routing
 all: $(CONTIKI_PROJECT)

MODULES += core/net/ipv4/uaodv
MODULES += core/net/ipv4/uaodv-rt core/net/rime

ifdef WITH_COMPOWER
APPS+=powertrace
CFLAGS+= -DCONTIKIMAC_CONF_COMPOWER=1 -DWITH_COMPOWER=1 -DQUEUEBUF_CONF_NUM=4
endif

 #UIP_CONF_IPV6=1
 CONTIKI_WITH_RIME = 1	CFLAGS =-DUIP_CONF_IPV6
MODULES += core/net/ipv6/multicast

 CFLAGS += -ffunction-sections
 LDFLAGS += -Wl,--gc-sections,--undefined=_reset_vector__,--undefined=InterruptVectors,--undefined=_copy_data_init__,--undefined=_clear_bss_init__,--undefined=_end_of_init__

ifeq ($(TARGET),avr-raven)
  COFFEE_FILES = 4
endif 

 CONTIKI =../../../
 include $(CONTIKI)/Makefile.include`
am I doing something wrong? I would be really appreciate if you answer me.