contiki-os/contiki #955: File overwrite? Intended behavior?

mkschreder, 2015-02-05:
When I tried opening an existing coffee fs file and writing to it, I had flash data being written to the same location (obviously resulting in garbage). Why is coffee fs not writing a new page when trying to overwrite an existing file? Do I always have to delete a file first when I need to overwrite it?

nvt, 2015-02-06:
It depends on the platform's configuration; Coffee should not overwrite flash data if COFFEE_MICRO_LOGS is set to 1 in cfs-coffee-arch.h.

mkschreder, 2015-02-10:
Ok that worked. Should be the default. It was very surprising to see file data corrupted when opening an existing file for writing..

nvt, 2015-02-17:
The default in Coffee is that COFFEE_MICRO_LOGS is 1. Among current Contiki platforms, avr-raven and native are the ones that set it to 0.

closed 2015-02-17 by nvt