contiki-os/contiki #972: Cannot modify make variables for travis compile tests

g-oikonomou, 2015-02-22:
For travis compile tests, it would be nice to be able to run:
```
make VARIABLE=something
```
rather than simply `make`.

Motivation: To use one example to test two (or more) different compile configurations for the same platform (e.g. 6LoWPAN fragmentation on/off, RPL on/off, RS/RA on/off, platform-specific options etc)

I would probably configure this within the job's Makefile.

I had a look at the various Makefiles of our testing framework and I couldn't figure out a straightforwar way to do this, nor could I think of a simple patch that could extend the framework to do something like that.

I wouldn't call this a bug, but certainly a limitation that IMHO needs addressed. Any ideas on how to patch things?

I don't want to go as far as calling this a bug, so I'll label it 'enhancement'

markushx, 2015-02-23:
Is
```make DEFINES=VARIABLE=something```
as mentioned in README-BUILDING.md working for you?

bthebaudeau, 2015-06-01:
Yes, `make DEFINES=VARIABLE=something` should work. There are even commands using it under `regression-tests/`.

The only exception is for the make files that already use it, because the command line variable would prevent the make file variable from working. In that case, the command line should use the value from the make file and add what's needed to it. Another solution would be to rewrite the make files using `DEFINE=`with `override DEFINE+=`, in which case the command line and the make file variables would be concatenated.

g-oikonomou, 2015-06-01:
So I've either articulated the question incorrectly, or I'm being very blind here. I am talking about getting travis to run a test with a specific value for a variable, not running the test myself in my PCs command line. As far as I can tell travis simply runs `make TARGET=x` for compile tests?

bthebaudeau, 2015-06-01:
If we take `regression-tests/Makefile.compile-test` and `regression-tests/18-compile-arm-ports/Makefile`, instead of putting a set of paths into the `EXAMPLES` variable, you should define a set of variables for each example, e.g.:
```
EXAMPLE_CC2538DK_DEFAULT_PATH=cc2538dk/cc2538dk
EXAMPLE_CC2538DK_DEFAULT_DEFINE=
EXAMPLE_CC2538DK_6LOWPAN_FRAG_ON_PATH=cc2538dk/cc2538dk
EXAMPLE_CC2538DK_6LOWPAN_FRAG_ON_DEFINE=SICSLOWPAN_CONF_FRAG=1
```

Then:
```
EXAMPLES+=EXAMPLE_CC2538DK_DEFAULT EXAMPLE_CC2538DK_6LOWPAN_FRAG_ON
```

And you change the make functions in `regression-tests/Makefile.compile-test` to work from that by appending `_PATH` and `_DEFINE` for each example.

g-oikonomou, 2015-11-20:
Feature added with #1327

closed 2015-11-20 by g-oikonomou