contiki-os/contiki #973: cc2538 RXFIFO underrun issues

areaarea, 2015-02-25:
I am having a bit of trouble getting the majority of my 'user space' packets through in contiki (~40% at best).  I am using the udp_send methods but I am losing packets to the lower layers.  

I turned on DEBUG in cc2538-rf.c (and limited the sheer volume of messages to those that were relevant to the issue) and I see the losses directly.  I have turned DMA on and off to make sure it wasn't an issue with the piping in any way.  The following logs are from when the DMA was disabled (last state I was in and is more verbose and relevant error messages WRT this issue)

An example of the error messaging around a message that is being 'lost':
--snip
RF: read (0x4c bytes) = 61dcc5cdab8d54a603004b12003351a603004b1200RF Error: 0x00000008
...
7aRF Error: 0x00000008 <-- One of these errors (RXFIFO underrun) per missing byte
...
317a <-- RSSI and CRC-OK
RF: Bad CRC 
RF: Init
RF: Set Channel
--/snip

The number of bytes that I get before error varies from 1 to a reasonable % of the packet, but the error state seems to be >50% (anecdotally from my testing and investigation)

When looking up the error bit I found (included inline):

--snip
TI users manual pp 682
23.10.2 Error Conditions
...
RX underflow is indicated when the RFERRF.RXUNDERF flag is set. RX underflow is a serious error
condition that should not occur in error-free software, and the RXUNDERF event should be used only for debugging or in a watchdog function. The RXUNDERF error is not generated when the read operation occurs simultaneously with the reception of a new byte.
--/snip

The manual states that the problem is a coding issue as the HW reports what it has in the buffer but then something must be interacting with the FIFO out of turn.

Has anyone else seen this behavior in practice? I have been seeing the down-stream effects (missing packets) for a long time (contiki 2.6 - Now) but have just got down to cleaning up the RF interactions (I was working on my own reliability layers which are easier to test on a less than stable target.)

Adam

areaarea, 2015-02-25:
For completeness, the printf error is generated in the fn:

cc2538_rf_err_isr()

g-oikonomou, 2015-02-25:
RXUNDERF is bad and, as you correctly point out, it should never happen. If it's caused by software, then it's a bug.

I must say that the last time I saw one was more than 2 years ago...

What are you using at the RDC layer?

areaarea, 2015-02-25:
Here is my boot POST:

Contiki-elarm-contiki-3.0-8-g53fdca5
TI SmartRF06 + cc2538EM
Net: sicslowpan
MAC: CSMA
RDC: ContikiMAC

We are experimenting with the timings of some of the out going packets on the sending nodes (less back to back) to ease congestion at the root with a small margin of success.

Has there been stress tests of the network?  I assume someone has done some metrics for actual throughput (vs. theoretical), message turn around times or some such for the actual hardware.   I am asking to clear up how well the radio support HW (CCA's, etc) work in practice.

Adam

g-oikonomou, 2015-02-25:
So when I saw UNDERF here is what was happening:

* Incoming packet, ISR polls the RF driver
* something reads the FIFO out
* the polled driver enters. Because it was polled, it thinks there is a packet in the FIFO (one that has been read out already): UNDERF.

A second scenario is:
* read() checks the first byte in the FIFO and it says that there is e.g. a 90-byte long frame in there. Then read starts reading out 90 bytes.
* Before it has finished it gets interrupted by something that, for whatever reason, clears/resets the FIFO.
* read() resumes and tries to read the remaining data, which are no longer there: UNDERF.

I think the current RF driver is clever enough to prevent/handle both of those situations.

I've not done any heavy stress-testing myself, but as I said earlier, things have so far been working solidly for me (incl. CCAs), but I've only used it in not very noisy environments.

areaarea, 2015-03-02:
All my own doing.  Nothing to do with repo. :)

closed 2015-03-02 by areaarea