contiki-os/contiki #975: er-rest-example on SmartRF06 board

serafimdasaudade, 2015-02-25:
Hi, 
Today I did git clone from the master branch, and try the er-example-server.
And the example didn't work. But is I run the same example, from the branch Contikie-2.7 runs OK.
But the output is strange:
Contiki-2.6-2048-g0c7d261
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: nullmac
 RDC: nullrdc
Rime configured with address 00:12:4b:00:04:13:32:6d

Why is Contiki-2.6 version.
Did I make the compilation wrong?

tks,

g-oikonomou, 2015-03-28:
I can see nothing wrong with the Contiki version string.

I am not sure what you mean by "the example didn't work". What's your setup? What behaviour did you expect? What did you observe?

g-oikonomou, 2015-05-09:
Inactive. Closing

closed 2015-05-09 by g-oikonomou