contiki-os/contiki #980: Problem with the travis build

marcas756, 2015-02-26:
Hello,

following builds failed:

https://travis-ci.org/contiki-os/contiki/builds/51772784
https://travis-ci.org/contiki-os/contiki/builds/52170020
https://travis-ci.org/contiki-os/contiki/builds/52175890

@ BUILD_TYPE='rpl' 	linux

I went through 52175895 and assume the reason to be:

```
Rearranging network
==== Files used for simulation (sha1sum) ====

xargs: unmatched single quote; by default quotes are special to xargs unless you use the -0 option

FAIL ಠ_ಠ

Running test 08-rpl-dao-route-loss-0 with random Seed 1: ................... OK

Running test 08-rpl-dao-route-loss-1 with random Seed 1: ................./../regression-tests/simexec.sh: line 70: 328 Aborted java -Xshare:on -jar $CONTIKI/tools/cooja/dist/cooja.jar -nogui=$CSC -contiki=$CONTIKI -random-seed=$RANDOMSEED > $BASENAME.log
```
Can someone have a look on it (I am not familiar with regression testing in contiki).

Regards Marcas

g-oikonomou, 2015-02-26:
Something wrong with that job, yes. Lost track of how many times I've had to restart it

marcas756, 2015-02-27:
FAIL ಠ_ಠ

No emoticons in code please

marcas756, 2015-03-18:
No further fails in the meantime ... seems to be solved somehow ... close

g-oikonomou, 2015-03-23:
So the build failed on master after the recent merge of #990 and I had to restart job 5. I'm re-opening this for further investigation.

Always good to have authors close their issues when they no longer need to be open by the way! (even though in this case it seems we may have jumped the gun :()

marcas756, 2015-03-29:
Currently there seems also to be a connectivity issue with travis (fails to download):
https://travis-ci.org/contiki-os/contiki/jobs/56321513

sumanpanchal, 2015-04-21:
Problem is still persist. Any suggestion please.

g-oikonomou, 2016-04-06:
I have not seen this occur in ages. Closing

closed 2016-04-06 by g-oikonomou