contiki-os/contiki #982: char is unsigned on BOARD=native on Linux ARM

gebart, 2015-03-08:
`char` is unsigned on ARM by default.

This causes trouble at multiple locations where a char return value is compared to a negative integer, e.g. `examples/ipv6/native-border-router` where the return value from getopt is compared against the numeric constant `-1` and the loop ends up in the default case when the end of parameters is reached instead of moving on.

Adding `-fsigned-char` to CFLAGS changes the behaviour to match the behaviour of `native` on x86_64 Linux

g-oikonomou, 2015-03-08:
Or we can always change the code to stop assuming signedness for the char datatype :)

gebart, 2015-03-08:
There are some pretty deeply rooted assumptions about char signedness. In one of the libc implementations or compiler builtins, I forgot which one, probably clang, there was `typedef char int8_t` even though it is specified that `int8_t` must be a signed type (That was how I learned that the default behaviour is different on ARM compared to x86).

g-oikonomou, 2015-03-09:
Speaking specifically about the example you brought up, the main problem is that the code in question uses a `char` to store the return value of `getopt`. All `getopt`s that I'm aware of (POSIX, as well as GNU _long extensions) return an `int`. IMHO that's the main problem here and I consider this to be a bug in the code, one that has nothing to do with `char` signedness.

> This causes trouble at multiple locations where a char return value is compared to a negative integer,

If you have found more such occurrences in platform-independent code, by all means report them.

More generally speaking, `char` signedness is undefined. In the interest of portability, platform-independent code should treat it as such. Each different implementation will have its own reasons for its choice about whether a char is treated as signed or unsigned:
* There are performance-related reasons why many (most? all?) compilers for x86 choose default: signed
* There are (or at least historically used to be) good reasons why the choice is default: unsigned on ARMs.
* SDCC has chosen default: signed for all supported targets and this has led to [some discussions](http://sourceforge.net/p/sdcc/mailman/message/32900834/) about how bad a decision this was (not sure whether they've changed it).

I will certainly not sign-off changes to build arguments in an attempt to harmonise `char` signedness across different Contiki platforms. IMHO this would be the wrong approach.