contiki-os/contiki #991: hello world example not working on cc2538

TobyEalden, 2015-03-12:
I'm a new to contiki so may be missing something obvious, but using the latest master the hello-world (nor any other) example does not work on the cc2538dk board.

I get the following output:

Contiki-2.6-2071-gc169b3e
TI SmartRF06 + cc2538EM
[hangs...]

Switching to the release-2-7 branch and everything works as expected.

g-oikonomou, 2015-03-12:
Hello Toby.

Just one merge ahead of you:

```
Contiki-2.6-2073-gc76316a
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: ContikiMAC
Rime configured with address 00:12:4b:00:ae:26:07:01
Hello, world
```

What OS are you building on? What toolchain version are you using? I'm on OS X with:
```
$ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/local/gcc-arm-none-eabi-4_9-2014q4/bin/../lib/gcc/arm-none-eabi/4.9.3/lto-wrapper
Target: arm-none-eabi
[...]
Thread model: single
gcc version 4.9.3 20141119 (release) [ARM/embedded-4_9-branch revision 218278] (GNU Tools for ARM Embedded Processors) 
```

TobyEalden, 2015-03-12:
I'm on Linux Mint 17 32-bit, kernel 3.13.0-24-generic, and toolchain:

Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/local/gcc-arm-none-eabi-4_8-2014q3/bin/../lib/gcc/arm-none-eabi/4.8.4/lto-wrapper
Target: arm-none-eabi
...
Thread model: single
gcc version 4.8.4 20140725 (release) [ARM/embedded-4_8-branch revision
213147] (GNU Tools for ARM Embedded Processors)

g-oikonomou, 2015-03-12:
Just tried with your toolchain version (still on OS X) - again no issues.

Can you try changing the value of `SMALL` between 0 and 1 either in the hello_example's `Makefile` or in `Makefile.cc2538dk`? You will have to make clean after each change.

A little bit of a strange question perhaps, but does your EM have an external antenna?

What version of the chip on your EM? (There should be a sticker at the back saying PG2 or somethng like that)

TobyEalden, 2015-03-12:
I changed SMALL from 1 to 0 in Makefile.cc2538dk and rebuilt and it is now
working...

What does that indicate?

g-oikonomou, 2015-03-12:
The optimiser will enable/disable different optimisations (and thus generate different code) between the two cases: `SMALL=1` optimises for size `-Os`, wheras `SMALL=0` optimises with `-O2`

More info #549 and #503

What I fail to get is how both versions boot on my EM, but one of them does not boot on yours...

If you can be bothered, I'd be interested to know if you observe the same situation for the two values of `SMALL` if you upgrade to a more recent toolchain version (e.g 4.9.x)

Just as I thought that at least _that one_ had been fixed for good :)

TobyEalden, 2015-03-13:
Just upgraded the toolchain to 4.9.3 20141119 to no avail - still fails if
SMALL = 1, and fine with SMALL = 0

FWIW, the sticker on the back of the EM says:

Texas Instruments
Model: CC2538EM
(DK) Rev.: 1.2.0.1
SN: 0000 0EBA

There is also a smaller sticker:

NO
1421
PbF

g-oikonomou, 2015-03-13:
So your EM doesn't have an external antenna, right? I've got EMs from an older batch: Mine says 1_0_1 and has an external antenna connector. The chip is identical though, as far as I know anyway.

I wonder...

TobyEalden, 2015-03-13:
Correct, there is no external antenna...

g-oikonomou, 2015-03-13:
I'll have a chat with TI then and I'll also see if I can grab some of those newer ones. I'm fairly certain a friend of mine nearby has a couple.

If anyone else has the same EMs as those described by Toby, it would be nice to get some success/failure reports.

Cheers

g-oikonomou, 2015-03-13:
So I just had a chat with various people who cannot reproduce it either, but one of them said something in the lines of:

"But he should be getting some of the netstack printfs"

Which made a certain degree of sense...

Try getting rid of `random_init()` please and let me know what happens. Also, it would help if you emailed me your broken .bin and the corresponding .map file. You have my email :)

Cheers

TobyEalden, 2015-03-13:
Commenting out random_init() in contiki-main.c made no difference...

g-oikonomou, 2015-03-15:
Turns out this was related with Uniflash / Toby's local setup.

closed 2015-03-15 by g-oikonomou