contiki-os/contiki #998: Use Additive Offsets for CC2538 Register Access

g-oikonomou, 2015-03-23:
Fork from #995 : 
> Also, using `REG(regs->base + OFFSET)` instead of the current `REG(regs->base | OFFSET)` found in the current CC2538 code base (unrelated to your PR) should generate more optimal assembly code. `|` is broken by design if the base is not aligned enough for the offset, and CPU instructions support addressing modes using additive offsets only (not OR-ed offsets), so computing a OR for each address used in the IP address range must be more expensive than the standard base `+` offset (because the compiler does not know the alignment of the variable base, so it can not optimize the | into a `+`), except if base is a constant known at compile time. This should be changed in the UART code (and everywhere else it appears), and the SPI code is OK here.

http://infocenter.arm.com/help/topic/com.arm.doc.dui0552a/BABJGHFJ.html

So the UART code is the obvious candidate. Where else do we need this?

bthebaudeau, 2015-03-23:
Yes, UART obviously, then this can be extended to the whole CC2538 code base as a coding rule, even if most other instances (if any) are probably compile-time constants, in which case the generated binary will be unchanged.

g-oikonomou, 2015-03-28:
#1009

closed 2015-03-28 by g-oikonomou