contiki-os/contiki #1006: Tunslip on Raspberry PI

latapata, 2015-03-26:
Hello All,

I want to connect CC2358dk RPL border router to Raspberry Pi via UART. Inside Raspberry PI I want to run tunslip6 utility. 

I just want to replace the instant-contiki with a Raspbian.

 Is this possible ?

dmazzer, 2015-03-27:
Hi,

I have run tunslip6 on a BeagleBone without any problem.
You will need to enable network tunneling protocols in kernel.

Regards,

_______________ _____ _____ __ ___ __ _ _ _  _
Daniel Mazzer
dmazzer@gmail.com
skype: danielmazzer

latapata, 2015-04-17:
I did not use Raspeberry Pi instead I started using a live Ubuntu booted from USB and running the tunslip on a computer to connect my border router.

Thanks for your answer @dmazzer

closed 2015-04-17 by latapata