contiki-os/contiki #1010: Native Border Router with slip on cc2538

Devesh24, 2015-03-28:
Hello Guys,
                   I am trying to make a topology where i am running slip-radio on cc2538 and native border router on linux machine, i have a cc2538 mote where i am running coap-server in which there are two observable resources, when i start observing both at the same time , after some time packet loss increases and even they stop responding. another problem is that if i observe one resource and from the linux machine i started the ping6 < Ip of mote > so they start interrupting each other and packet loss increases.
Any lead ?
Thanks.

g-oikonomou, 2015-03-28:
Davesh, I'm not exactly sure what you were expecting to see while this ping is running? I'd expect increased traffic to increase packet loss, especially under ContikiMAC.

Devesh24, 2015-03-30:
@g-oikonomou 
Thanks for reply, i got your point.
But there is still a problem , if i observe 2 resources at a time on same coap server then after some time like 20 min it stopped to getiing data on copper, from the log i can see that after some time border-router doesn't send ACK for con message so coap server dropped client as a subscriber while coap server is still running without any error or buffer overflow just stop sending data.
I fetched commit that implemented the circular buffer #920 , but same problem.

g-oikonomou, 2015-04-05:
What's the configuration for the SLIP radio in terms of UART vs USB? Have you made any other configuration changes?

Devesh8, 2015-04-06:
Right now i am using slip over UART only no changes made in conf.

Devesh24, 2015-04-06:
Right now i am using slip over UART only, no other changes made .

closed 2015-05-09 by g-oikonomou