contiki-os/contiki #1014: Which ARM based board for a Duty Cycling MAC ?

Cnadir91, 2015-04-01:
Hi everyone, 

I used to work with Contiki OS on the famous Tmote-sky (MSP based) platform. 
For now, I want to use Contiki OS on an ARM based Platform. 

After a search, i found that Contiki was ported (completely or partially) ported these boards :
    - Econotag
    - TI CC2538
    - stm32fw
    - M3 Node (On IoT-Lab i thing it's not possible to buy it)

Finally, i am really interested in the Econotag board but after some reading, i found that this board does not really support well, a Duty Cycling MAC protocol. 

My questions are : 
    i) Does the Contiki main line or a fork give a full RDC protocol support for the Econotag board ?
    ii) Which are the other ARM based boards, that could give me a full support for a RDC Protocol (X-mac or ContikiMAC) ? 

Thank you in advance for your advises.
Nadir.

gebart, 2015-04-01:
We have a board called Mulle which is powered by an ARM Cortex-M4 and using the AT86RF212B radio which we are using in Contiki with ContikiMAC for low power operation as well as other power management methods for battery powered sensor applications. Contact me at info@eistec.se if you are interested in hearing more about this.

g-oikonomou, 2015-04-05:
CC2538 and CC26xx have full ContikiMAC support. Both CM3

closed 2015-04-30 by g-oikonomou