contiki-os/contiki #1018: Improve security of AES software implementation

david-oswald, 2015-04-07:
Hi,

in it's current form the AES implementation (https://github.com/contiki-os/contiki/blob/master/core/lib/aes-128.c) appears to be vulnerable to timing attacks due to the use of data-dependent (and hence also key-dependent) branches in galois_mul2(). 

Same holds for cache timing (if the underlying processor has a cache of course, not sure if this is the case for any of the supported platforms): http://cr.yp.to/antiforgery/cachetiming-20050414.pdf

Best
Davud

kkrentz, 2015-05-15:
Good point. We could build some decorater that conceals the execution time. Do you know if hardware-accelerated AES implementations are vulnerable to such side channel attacks, too?

david-oswald, 2015-05-15:
Not fully sure about the decorator - do you mean "insert appropriate waiting at the end" or something alike? I'd rather go for a constant-time implementation, e.g. T-tables if on 32-bit or for 8-bit realizing xtime() with boolean operations, see e.g. https://github.com/kokke/tiny-AES128-C/blob/master/aes.c

This would certainly make things better, though to be sure one would always have to check the generated assembly or measure the actual runtime.

HW implementations are usually constant-time (e.g. on AVR, MSP, ...), so using these is advisable (also since they are *much* faster). Of course we are only talking about the timing side channel, other attacks (e.g. power or EM emanation are still possible). Protecting against these is a different challenge, but at least they require physical access, while a timing attack could be possible over the RF interface.