contiki-os/contiki #1023: CC2538 + CC2592 driver

dazhbog, 2015-04-10:
Hi, Im trying to get the CC2538-CC2592emk (TI black boards) to work.

Looking at the manual one must set some pins, and 4 registers.

The registers are shown in this appnote.
[image]

I use the following code in cc2538-rf.c (in the init function) to set the registers. 
```
  REG (RFCORE_XREG_RFC_OBS_CTRL0) = 0x6A;
  REG (RFCORE_XREG_RFC_OBS_CTRL1) = 0x68;
  REG (CCTEST_OBSSEL2) = 0x80;
  REG (CCTEST_OBSSEL3) = 0x81;
```
The CCTEST_OBSSEL2 declarations seem to be in cctest.h which is not included in the rfcore.h. Why is that?

I am also using the following code
```C
  #define HGM 		(1 << 2) //PD2
  GPIO_SET_OUTPUT(GPIO_D_BASE, HGM);		//HGM
  GPIO_SET_PIN(GPIO_D_BASE, HGM);
```
To set the thing in high gain.

I have LEDs tied to the pins to see if there is any activity, for now i cant see anything. I will try the range soon, but for now, am I doing something wrong here?

g-oikonomou, 2015-04-11:
PC2 and PC3 are also used by the Srf LED driver. I had a quick look at the UG and I _think_ that `CCTEST_OBSSEL[0..7]` will override GPIO behaviour, but perhaps something to look at.

> The CCTEST_OBSSEL2 declarations seem to be in cctest.h which is not included in the rfcore.h. Why is that?

`cctest.h` appeared in the repo after the RF driver, which at the point of writing didn't need it.

The UG (Sec. 9.2.4 - Radio Test Output Signals) mentions something about setting pullup on the pin. Not sure whether this is needed for the CC2592, but perhaps another thing to look into.

jonnteolsson, 2015-04-14:
Suggest to post on e2e.ti.com

retfie, 2015-04-15:
You can reuse this patch
https://github.com/retfie/contiki/commit/c6ca74d8740b9b41b0dbf7a44e7abd74fed0704b
. It's for CC2591 and still experimental , but works very well for me for
more than several months.

>
> Hi, Im trying to get the CC2538-CC2592emk (TI black boards) to work.
>
> Looking at the manual one must set some pins, and 4 registers.
>
> The registers are shown in this appnote.
>
> I use the following code in cc2538-rf.c (in the init function) to set the
registers.
>
>   REG (RFCORE_XREG_RFC_OBS_CTRL0) = 0x6A;
>   REG (RFCORE_XREG_RFC_OBS_CTRL1) = 0x68;
>   REG (CCTEST_OBSSEL2) = 0x80;
>   REG (CCTEST_OBSSEL3) = 0x81;
>
> The CCTEST_OBSSEL2 declarations seem to be in cctest.h which is not
included in the rfcore.h. Why is that?
>
> I am also using the following code
>
>   #define HGM       (1 << 2) //PD2
>   GPIO_SET_OUTPUT(GPIO_D_BASE, HGM);        //HGM
>   GPIO_SET_PIN(GPIO_D_BASE, HGM);
>
> To set the thing in high gain.
>
> I have LEDs tied to the pins to see if there is any activity, for now i
cant see anything. I will try the range soon, but for now, am I doing
something wrong here?

dazhbog, 2015-04-16:
I tried your patch @retfie and I think it works. It went from an RSSI of -60/-70dbi to a -20dbi when having the boards side by side. I did a preliminary range test of up to 300 meters and it was working.

Why did you place your code in the on() function? Do the registers lose their value (or some state is lost) after a PM2 sleep?

I think my code and yours was the same apart from the placement.

Thanks again!

closed 2015-04-30 by g-oikonomou

edilcan, 2015-10-22:
@dazhbog 
where is the place of code implementing CC2592 patch?
Because I have tested it in the on() function and it does not work?

marketideas, 2016-07-22:
Shouldn't this code actually be placed in the board and/or platform files?  The LNA control pins are external to the chip and the PCB determines which pins between the CC2538 and the CC2592 are connected.  On my board (custom) PC5 and PC6 are used as the CC2592 control pins.

marcomilazzo, 2016-12-23:
can you tell me  were to put thecode ?
marco

dazhbog, 2016-12-31:
@edilcan @marcomilazzo 

Its been a while to touch this piece of code and even use these boards.. All i did is I put the following code in the on() function, in cpu/cc2538/dev/cc2538-rf.c, right before the ENERGEST call. 
```C
/* RF observable control register value to output PA signal */
#define RFC_OBS_CTRL_PA_PD_INV        0x6A
/* RF observable control register value to output LNA signal */
#define RFC_OBS_CTRL_LNAMIX_PD_INV    0x68 
// RF observable control register value to output LNA signalfor CC2591 compression workaround.
#define RFC_OBS_CTRL_DEMOD_CCA        0x0D
/* OBSSELn register value to select RF observable 0 */
#define OBSSEL_OBS_CTRL0             0x81
/* OBSSELn register value to select RF observable 1 */
#define OBSSEL_OBS_CTRL1             0x80
#define RFC_OBS_CTRL0 REG(RFCORE_XREG_RFC_OBS_CTRL0)
#define RFC_OBS_CTRL1 REG(RFCORE_XREG_RFC_OBS_CTRL1)
#define OBSSEL3   REG(CCTEST_OBSSEL3)
#define OBSSEL2   REG(CCTEST_OBSSEL2)
#define EN_LNA_PIN  (1 << 2)
#define PAEN_PIN    (1 << 3)
#define HGM_PIN     (1 << 0)

  GPIO_SET_OUTPUT(GPIO_C_BASE, PAEN_PIN);
  GPIO_WRITE_PIN(GPIO_C_BASE, PAEN_PIN, 1);
  GPIO_SET_OUTPUT(GPIO_C_BASE, EN_LNA_PIN);
  GPIO_WRITE_PIN(GPIO_C_BASE, EN_LNA_PIN, 1);
  // PD0 -> 0/1 -> High/Low Gain
  GPIO_SET_OUTPUT(GPIO_D_BASE, HGM_PIN);
  GPIO_WRITE_PIN(GPIO_D_BASE, HGM_PIN, 1);
  // PC3 -> PAEN
  RFC_OBS_CTRL0 = RFC_OBS_CTRL_PA_PD_INV;
  OBSSEL3 = OBSSEL_OBS_CTRL0;
  // PC2 -> EN (LNA control)
  RFC_OBS_CTRL1 = RFC_OBS_CTRL_LNAMIX_PD_INV;
  OBSSEL2 = OBSSEL_OBS_CTRL1;
  //
```
Also added this include at the top of cc2538-rf.c
```C
#include "dev/cctest.h"
```

This is the same as seen at retfies repo  [retfie@c6ca74d](https://github.com/retfie/contiki/commit/c6ca74d8740b9b41b0dbf7a44e7abd74fed0704b)

Hope this helps.

**Also I have a alot of these black boards (around 20) if anyone is interested to get them for cheap PM me.**

marcomilazzo, 2017-02-13:
hi
can you please tell  me  how to change the pins assigment?

immagine i want to use port c or whatever
thank's a lot
marco

marcomilazzo, 2017-06-18:
hi
any code for the cc2592 and cc2560 ?
thank's
marco