contiki-os/contiki #1025: CC2538 getting RSSI value

dazhbog, 2015-04-11:
Hi all,

I am trying to get the RSSI value from the CC2538.

When run the following code into the receiving function I get
 **RSSI: -104 dBm [0xff98]** sometimes  **RSSI: -105 dBm [0xff97]**
which doesn't seem to change when I move the sensor. 

**Code**
```C
printf("RSSI: ");
  if(get_param(RADIO_PARAM_RSSI, &value) == RADIO_RESULT_OK) {
    printf("%d dBm [0x%04x]\n", value, (uint16_t)value);
  }

//UTIL
static radio_result_t
get_param(radio_param_t param, radio_value_t *value)
{
  radio_result_t rv;

  rv = NETSTACK_RADIO.get_value(param, value);

  switch(rv) {
  case RADIO_RESULT_ERROR:
    printf("Radio returned an error\n");
    break;
  case RADIO_RESULT_INVALID_VALUE:
    printf("Value %d is invalid\n", *value);
    break;
  case RADIO_RESULT_NOT_SUPPORTED:
    printf("Param %u not supported\n", param);
    break;
  case RADIO_RESULT_OK:
    break;
  default:
    printf("Unknown return value\n");
    break;
  }
  return rv;
}
```

**[EDIT]**
I tried the following code and it seems to change now, not sure if its still a reliable way of reading the RSSI value.

**Code**
```C
printf("last_rssi %d\n",(signed short)packetbuf_attr(PACKETBUF_ATTR_RSSI));
```

Any input on the correct way of obtaining the RSSI value would be much appreciated. Thanks!

g-oikonomou, 2015-04-11:
When the driver reads out a received frame, the RSSI observed during reception will get copied to the packetbuf and this is what you are getting by calling `packetbuf_attr()`

`get_param(RADIO_PARAM_RSSI, &value)` will basically read the value of `RFCORE_XREG_RSSI`. If the RF driver is off at the time you make the call, it will turn on and wait for the RSSI valid signal. If there is traffic in the air, I bet you will see higher values.

Which method to use mainly depends on the point in time you need RSSI for: whether you need the RSSI value _now_, or during last frame reception.

dazhbog, 2015-04-11:
Ok so if I understand correctly, If I want the RSSI value every time I receive a packet, in the tcp handler, I need to use the `packetbuf_attr()` ?

g-oikonomou, 2015-04-11:
Yes, bearing in mind that the packetbuff attribute will change every time you receive a packet (including e.g. routing, ND control packets), so that includes packets your process will never actually see. In the approach that you are describing, you will only be reading RSSI for packets that were actually delivered to your process.

g-oikonomou, 2015-04-11:
A couple more thoughts:

* There is also `sicslowpan_get_last_rssi()`
* Another thing to be careful of is 6lowpan fragmentation. If your process sees a datagram which has undergone reassembly, the RSSI will correspond to that observed during reception of the last fragment, which may or may not be what you need, again depending on what you are trying to achieve.

dazhbog, 2015-04-13:
Excelent, thanks George!

I think for what I need, reading the packet's RSSI that only arrives in my process is more than enough.

closed 2015-04-13 by dazhbog

sepcos, 2017-04-11:
hi every one i new in contiki and i need your help

sepcos, 2017-04-11:
i need to mesaure rssi tx power code? as min and max