contiki-os/contiki #1035: Application Layer / Architecture including OTA, etc. for Contiki (interest poll)

joakimeriksson, 2015-04-24:
I have been working quite a while at Yanzi networks developing Contiki software and now we are considering to upstream some of the application layer software and extensions that have been developed. My question is if there are any interests in a complete application architecture implementation in upstream Contiki?

It includes support for
- Discovery of functionality of devices
- OTA for full firmware image upgrade
- Read/write variables in the application object / instance model.

This application architecture have been ported to STM32W and CC2538 platforms
(and also tested on smaller platforms) and is based on a UDP protocol.

Included in this is also a set of scripts, libraries and tools for doing discovery, OTA, etc. And full documentation of the protocol and application architecture.

jbnhf23, 2015-04-25:
indeed. Could be awesome. Reinventing the wheel, is not what I prefer ;-)

areaarea, 2015-04-29:
That would be awesome!

Adam

joakimeriksson, 2015-05-06:
Ok, so thanks for the thumbs - I will start lobbying a bit to get this into shape for open-sourcing then. We are also working on some other cool and useful things that I will try to get upstreamed!

sdawans, 2015-05-11:
:+1:

I would try to make OTA as modular as possible - the update logic and underlying code dissemination strategies are vast

jonnteolsson, 2015-05-12:
Utilizing this and an open standards like OMA LWM2M on top of COAP would be really nice.

joakimeriksson, 2015-05-12:
Totally agree! But the first version (the one we already have) will be using another very similar approach (almost the same, but before OMA LWM2M suggested their object model). And I guess LWM2M is still in some need for getting completely final before implementing it fully.

areaarea, 2015-05-12:
My druthers would be to get an OTA option in the tree but to make sure it is factored in a way that accommodates logical upgrades that we can see coming now (e.g. OMA LWM2M) and normal separations (network layers and such).  But getting something in place that can be improved would be a huge step forward IMO.

troymccann, 2015-10-06:
Are there any updates for this? I'm looking into OTA firmware updates with the CC2538 with Contiki at the moment; it would be great to avoid reinventing the wheel!

joakimeriksson, 2015-10-07:
Yes, this is work in progress - we are preparing a pull request for the application layer soon. With luck it will be a PR later this week.

troymccann, 2015-10-07:
Excellent, thanks @joakimeriksson ! Very much appreciated. :)

MaximilianWorthZeroMillion, 2015-10-30:
Hey @joakimeriksson,
your work is highly appreciated! What is your current status?

joakimeriksson, 2015-10-30:
Sorry for being a bit slow. We are preparing the PR's for this right now - but also a bit depend on the new 6LoWPAN adaptation layer PR is getting in there. But we will try to make these PR's that depend on that PR one if it is not pulled anytime soon.

jonnteolsson, 2015-11-22:
any updates, we see great interest in such a feature!

joakimeriksson, 2015-11-22:
Push any of the mergers to pull the 6LoWPAN adaptation layer PR and we can try to upstream this ASAP ;-)

JelmerT, 2015-11-24:
Looking forward to this one :+1:

knowic, 2016-01-22:
Thanks，we are looking forward to it
....

joakimeriksson, 2016-01-22:
Yes. Now when the adaptation layer is in we will start pushing this soon!

tonnenpinguin, 2016-02-29:
Any updates on this @joakimeriksson? ;-)

joakimeriksson, 2016-02-29:
We are working on it - we might start pointing you to our "work-in-progress" branch so that you could do early tests. Sorry for being slow on this - I have spent a lot of time fixing other issues. But I it on the top-list of our priorities.

tonnenpinguin, 2016-03-01:
Thanks for the update!
Of course we could contribute some testing/bug reporting!

lligios, 2016-03-03:
Thank you @joakimeriksson 
We could also contribute for testing and development.

msolters, 2016-03-20:
Anything testable yet?

knowic, 2016-04-11:
Is there any clear time budget about this？ We are looking forward it.

joakimeriksson, 2016-04-19:
No, clear time budget - but we will work on this in a few days - hopefully able to get a testable version for anyone to test before the weekend!

LanshunNie, 2016-05-26:
Is there any update? We are looking forward to OTA, : )

joakimeriksson, 2016-05-26:
Yes, we are working on it - got it to a level where we have one platform supported (a CC2538 based) but are adding Zolertia Re-Mote now so that there is an official Contiki platform to test with also. We will work with it today but I do not expect it to be fully ready today.

LanshunNie, 2016-05-26:
Awesome!

joakimeriksson, 2016-08-16:
Ok, anyone who like to test can find the pre-release of the Sparrow application layer and tools here:
https://github.com/sics-iot/sparrow

Includes bootloader, OTA-tools, new NBR / Serial Radio, etc.