contiki-os/contiki #1038: Contiki IPv6 node remote access over IPv4 network

kashifsanadar, 2015-04-28:
Hi,
I want to access IPv6 Contiki network nodes from a remote computer via IPv4 based Internet. I noticed the latest development of IP64 module under Contiki "/core/net/ip64" but do not see any example code demonstrating its full potential. My question is "Is it possible to poll an IPv6 enabled Contiki device from an IPv4 enabled remote computer using e.g., IP64 module?" If yes, do we have sample code for that? If no, what is other possible solution?

FYI, I am using the following testbed,
A Tmote sky node running RPL border router and  attached with a Ubuntu machine via USB
Other Tmote sky Contiki nodes running "rest-server" example code
Remote Windows/Linux machine making HTTP GET request to one of the Tmote sky node

Thanks in advance!

Kashif

mcr, 2015-04-28:
kashifsanadar <notifications@github.com> wrote:
    > I want to access IPv6 Contiki network nodes from a remote computer via
    > IPv4 based Internet. I noticed the latest development of IP64 module
    > under Contiki "/core/net/ip64" but do not see any example code
    > demonstrating its full potential. My question is "Is it possible to
    > poll an IPv6 enabled Contiki device from an IPv4 enabled remote
    > computer using e.g., IP64 module?" If yes, do we have sample code for
    > that? If no, what is other possible solution?

0) You can not run layer-3 packets that way without either a tunnel to get IPv6,
   or some kind of application layer gateway.

1) Your IPv6 connected motes can use NAT64 to connect outbound to your IPv4
   server.

2) You could deploy an IPv4 capable CoAP proxy. Could be a CoAP/HTTP proxy.

Either way, you need a device with an IPv4 address and an IPv6 address either
to manage a tunnel, translate or proxy.

vsaw, 2015-04-29:
Have you seen the [`ip64-special-ports`](https://github.com/contiki-os/contiki/blob/master/core/net/ip64/ip64-special-ports.h)? It allows the IP64 node to forward certain packets arriving at a specific IPv4 port into the IPv6 network.

alignan, 2015-05-17:
Closing, questions should be addressed to the mailing list.

closed 2015-05-17 by alignan