contiki-os/contiki #1049: Use of mac_call_sent_callback in nullrdc.c

aguirrem, 2015-05-06:
I sent this question to the mailing list. I am posting it here in hopes to get a bit more visiblity

Looking at the code in nullrdc.c I noticed that the last parameter in mac_call_sent_callback (numtx) is always set to '1'.

By looking at the prototype of the function mac_call_sent_callback:
```
mac_call_sent_callback(mac_callback_t sent, void *ptr, int status, int num_tx)
```

From the name one can deduce that the value expected in *num_tx* is the number of attempts that the packet took to be transmitted by the radio. This value is eventually passed down to the the configured RPL objective function. However with nullrdc that numtx will always be 1 even though the number of attempts can easily be retrieved from the radio driver.

My questions are the following:
 * Why is this value fixed at "1" currently?
 * Would there ever be a reason to use the true number of tx attempts? if so when?

Thanks in advance

laurentderu, 2015-05-07:
The num_tx parameter makes sense only for CSMA as it represents the number of actual transmission performed. The radio driver and the RDC will never try to retransmit a packet after a failure, hence the value is fixed to 1. Only in CSMA you have retransmission and so different values for num_tx.

aguirrem, 2015-05-07:
maybe I am misunderstanding the concept but the radio hardware itself supports retransmission when a hardware acknowledgment is not received even without CSMA does it not? 
Or is the num_tx argument specifically meant for soft-retransmissions?

g-oikonomou, 2015-05-09:
AFAIK most radios (at least the ones I've played with) have a capability to wait for ACKs and to retransmit, but most radios supported by Contiki have this feature turned off (I'm asking to be corrected here).

closed 2015-05-09 by g-oikonomou

aguirrem, 2015-05-10:
@g-oikonomou  
The reason why I am inquiring about this is because I am currently working on a port for SAMR21 (Cortex M0+) and I'd like to turn the auto-retrasmission on.  Is there a reason why this is turned off on the radios you have worked with? 

I don't mean to keep beating a dead horse but I still don't quite understand. is num_tx specifically intended for CSMA retries? or can it be used for hardware retries in your view does would it provide any information regarding the suitability of a parent node?

aignacio, 2016-03-05:
@aguirrem  are you still doing this PORT?

shankhadas, 2019-05-02:
Can you help on how to run mac.c to trace collision, no acknowledgement and sent packates between sink and sender motes in cooja simulator?