contiki-os/contiki #1058: Cortex M3 .hex files

g-oikonomou, 2015-05-09:
The CC2538 and CC26xx ports are identical in terms of the final .bin and .hex

```
arm-none-eabi-objcopy -O ihex cc2538-demo.elf cc2538-demo.hex
arm-none-eabi-objcopy -O binary --gap-fill 0xff cc2538-demo.elf cc2538-demo.bin
```

In a nutshell, we create a .elf and from it we create a .bin and a .hex

The `.bin` works perfectly with CC2538 as well as with CC26xx. The .hex works fine with CC2538, but it fails miserably with CC26xx. TI's flash prog complains about an unknown record type, but I suspect this is not the primary cause.

I ran objdump on both hex files

$ arm-none-eabi-objdump -h cc26xx-demo.hex 

cc26xx-demo.hex:     file format ihex

Sections:
Idx Name          Size      VMA               LMA               File off  Algn
  0 .sec1         0000d6bd  00000000  00000000  00000000  2**0
                  CONTENTS, ALLOC, LOAD
  1 .sec2         00000008  0000d6c0  0000d6c0  00025bf6  2**0
                  CONTENTS, ALLOC, LOAD
  2 .sec3         00000054  0001ffac  0001ffac  00025c24  2**0
                  CONTENTS, ALLOC, LOAD
```

and this looks normal to me.

Any ideas?

bthebaudeau, 2015-05-09:
You can first test if TI's flash prog is lying or not. Just use a text editor to remove all the records of each type, type after type, until TI's flash prog no longer complains:
http://en.wikipedia.org/wiki/Intel_HEX#Record_types

The record type 0x00 should be fine, but maybe some of the other record types are not correctly supported by TI's flash prog. Does it have a command line option to force the input file format? Is there a big difference between the flash address ranges of the CC2538/CC26xx?

g-oikonomou, 2015-05-09:
> Is there a big difference between the flash address ranges of the CC2538/CC26xx?

CC26xx starts at 0x0 (which is what Arm likes AFAIK), whereas CC2538 starts at 0x20000

> You can first test if TI's flash prog is lying or not.

This is the offending record type with CC26xx:

```
:04000003000000E910
```

whereas with the CC2538 we get

```
:020000040027D3
```

So objcopy ultimately gives us one record type for one platform, but it gives a different record type for the other. /stumped

g-oikonomou, 2015-05-09:
I should say, we changed the CC26xx record type for CC26xx (and the checksum accordingly), Flash Prog stopped complaining but progamming the .hex still did not work

bthebaudeau, 2015-05-10:
I wanted to take a look at it, but the `cc26xx-demo` example does not build using the current instructions and dependencies:
```
  CC        ../../platform/srf06-cc26xx/./contiki-main.c
../../platform/srf06-cc26xx/./contiki-main.c:64:37: fatal error: driverlib/driverlib_ver.h: No such file or directory
 #include "driverlib/driverlib_ver.h"
                                     ^
compilation terminated.
```

`driverlib/driverlib_ver.h` and the `DRIVERLIB_*_VER` definitions no longer exist (new release of cc26xxware on May 5). They seem to have been superseded by `driverlib/driverlib_release.h`. Older releases are not available for download, and anyway the documentation in Contiki does not say which version it is supposed to be compatible with.

Contiki should not rely on an external dependency out of control. The supported archives should be either made available for download on some servers controlled by Contiki (at least as a backup), or the source code they contain should be integrated into Contiki and regularly upgraded, or this could also be a Git submodule under the contiki-os organization. And all these operations could be automated for the build instead of having to manually download this dependency and set `TI_CC26XXWARE`.

bthebaudeau, 2015-05-10:
If you want to see what TI's flash prog does, you can program the `.bin`, then read a `.bin` and compare. Same with `.hex`/`.bin` (in the case it does not complain first). Then, compare the read `.bin` files.

bthebaudeau, 2015-05-10:
It's not the 0x04 record type that CC2538 is using instead of 0x03 on CC26xx, but 0x05. 0x04 is just used to program stuff at 0x20XXXX, then at 0x27XXXX. The issue with CC26xx is that it's using segmented addresses instead of linear addresses.

I've removed the offending version stuff, just to try building, but then it breaks everywhere, e.g.:
```
In file included from ../../platform/srf06-cc26xx/srf06/button-sensor.c:46:0:
../../platform/srf06-cc26xx/srf06/button-sensor.c: In function 'config_buttons':
../../platform/srf06-cc26xx/srf06/button-sensor.c:61:34: error: 'IOC_JTAG_DISABLE' undeclared (first use in this function)
                                  IOC_JTAG_DISABLE)
                                  ^
../../cpu/cc26xx/./ti-lib.h:256:66: note: in definition of macro 'ti_lib_ioc_port_configure_set'
 #define ti_lib_ioc_port_configure_set(...)   IOCPortConfigureSet(__VA_ARGS__)
                                                                  ^
../../platform/srf06-cc26xx/srf06/button-sensor.c:198:55: note: in expansion of macro 'BUTTON_GPIO_CFG'
     ti_lib_ioc_port_configure_set(key, IOC_PORT_GPIO, BUTTON_GPIO_CFG);
                                                       ^
../../platform/srf06-cc26xx/srf06/button-sensor.c:61:34: note: each undeclared identifier is reported only once for each function it appears in
                                  IOC_JTAG_DISABLE)
                                  ^
../../cpu/cc26xx/./ti-lib.h:256:66: note: in definition of macro 'ti_lib_ioc_port_configure_set'
 #define ti_lib_ioc_port_configure_set(...)   IOCPortConfigureSet(__VA_ARGS__)
                                                                  ^
../../platform/srf06-cc26xx/srf06/button-sensor.c:198:55: note: in expansion of macro 'BUTTON_GPIO_CFG'
     ti_lib_ioc_port_configure_set(key, IOC_PORT_GPIO, BUTTON_GPIO_CFG);
                                                       ^
../../Makefile.include:241: recipe for target 'obj_srf06-cc26xx/button-sensor.o' failed
```

Very incompatible new cc26xxware release...

g-oikonomou, 2015-05-10:
It's a fairly big patch needed to use the new cc26xxware. Currently we need this one: http://www.ti.com/litv/zip/swrc296

bthebaudeau, 2015-05-10:
`0xfffff` is the address threshold triggering the switch between the `0x02`/`0x03` and the `0x04`/`0x05` `ihex` record types in `objcopy`:
https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;a=blob;f=bfd/ihex.c;h=09f756a1c2c11e57c5da15a6ff96491275575b86;hb=237df3fa4a1d939e6fd1af0c3e5029a25a137310#l791
https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;a=blob;f=bfd/ihex.c;h=09f756a1c2c11e57c5da15a6ff96491275575b86;hb=237df3fa4a1d939e6fd1af0c3e5029a25a137310#l860

The record type is the 4th argument passed to `ihex_write_record()`:
https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;a=blob;f=bfd/ihex.c;h=09f756a1c2c11e57c5da15a6ff96491275575b86;hb=237df3fa4a1d939e6fd1af0c3e5029a25a137310#l712

No option passed to `objcopy` may help here. The generated `.hex` file is perfectly valid. The only issue is TI's flash prog, which does not support standard `ihex` files. So the choices are:
 - if TI's flash prog supports the `.bin` format:
    * program a uselessly big (because of the filled gap) `.bin` file,
    * split the `.bin` to skip the gap,
 - with the `.hex` format:
    * pass a global LMA offset to `objcopy` to put everything beyond `0xfffff`, then pass the opposite address offset to TI's flash prog if it supports such an option,
    * use a script or something to remove the type-`0x03` record since it should not be used by TI's flash prog, which requires that it works fine with type-`0x02` records,
    * use a script or something to transform the type-`0x02`/`0x03` records into types `0x04` and `0x05` (the addresses should be changed accordingly),
 - use another flash prog.

PR #1060 (2015-05-14): Produce ihex files usable with TI's Flash Programmer
As per the discussion in #1058, TI's Flash Programmer seems to dislike 03 records in ihex files. As a result, the `.hex` file generated by the CC26xx build system cannot be used with Flash Programmer.

This pull changes the CC26xx build system as follows:
* We `objcopy` from the `.elf` to an intermediate `.hex` ~~which has all addresses offset by `0x200000`. This forces `objcopy` to only use 04/05 records instead of 02/03.~~
* We use `srecord` to post-process this intermediate hex. This produces a final hex file that uses 32-bit linear addre [... cut]
closed 2015-05-14 by bthebaudeau