contiki-os/contiki #1061: TARGET=sky doesn't work with mspgcc-ti

flokli, 2015-05-10:
Hi,

I wanted to compile the examples outside the VM (in Arch Linux) for the 'sky' target.

In AUR, there are some mspgcc [packages from 2012](https://aur.archlinux.org/packages/gcc-msp430/), which are discontinued. The recommendation is to use the [mspgcc-ti packages from TI](https://aur.archlinux.org/packages/mspgcc-ti).

These bring a `msp430-elf-gcc` (not `msp430-gcc`).

Version `msp430-elf-gcc (GCC) 4.9.1 20140707 (prerelease (msp430-14r1-167)) (GNUPro 14r1) (Based on: GCC 4.8 GDB 7.7 Binutils 2.24 Newlib 2.1)`
However, the Makefile didn't find the binary, so I symlinked /opt/ti/mspgcc/bin/msp430-gcc to msp430-gcc.

During build, I now get the following error:

```
cd examples/hello-world
make TARGET=sky
  CC        ../../platform/sky/./contiki-sky-platform.c
In file included from ../../platform/sky/./platform-conf.h:68:0,
                 from ../../platform/sky/./contiki-conf.h:9,
                 from ../../core/./contiki.h:38,
                 from ../../core/./lib/sensors.h:36,
                 from ../../core/./dev/button-sensor.h:35,
                 from ../../platform/sky/./contiki-sky-platform.c:35:
../../cpu/msp430/./msp430def.h:48:16: fatal error: io.h: No such file or directory
 #include <io.h>
                ^
compilation terminated.
../../Makefile.include:241: recipe for target 'obj_sky/contiki-sky-platform.o' failed
make: *** [obj_sky/contiki-sky-platform.o] Error 1
```

Whats the proper way to build for the "sky" TARGET?

xwirth, 2015-05-20:
Hi,

mspgcc-i does not use the same librairies and headers as mspgcc of Debian, which is the 2012 version, and the one used by Contiki.
However you can use the binairies from Debian by installing this AUR packages : 
mspgcc-binutils-bin
mspgcc-gcc-bin
mspgcc-libc-bin
mspgcc-mcu-bin

You could have another issue in arch : the python version needed by Contiki is python2 but by default it calls python3. I had to modify the headers of the python files (#!/usr/bin/env python -> #!/usr/bin/env python2).

flokli, 2015-05-21:
Awesome, this worked for me!

I added the information to the contiki wiki: 
https://github.com/contiki-os/contiki/wiki/Setup-Contiki-Toolchain-in-Arch-Linux

closed 2015-05-21 by flokli