contiki-os/contiki #1068: Cooja Compile failed

xinqiu, 2015-05-13:
I use the method [Setup contiki toolchain in ubuntu](https://github.com/contiki-os/contiki/wiki/Setup-contiki-toolchain-in-ubuntu-13.04) to install contik.i

When I want to use the Coojia,there's something wrong.I don't know how to deal with it.

```
qiuxin@ubuntu:~/contiki/tools/cooja$ ant run
Buildfile: /home/qiuxin/contiki/tools/cooja/build.xml

init:

compile:
    [javac] Compiling 164 source files to /home/qiuxin/contiki/tools/cooja/build
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:419: illegal start of type
    [javac]     HashMap<String, Symbol> variables = new HashMap<>();
    [javac]                                                     ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:632: illegal start of type
    [javac]       Map<String, Symbol> varNames = new HashMap<>();
    [javac]                                                  ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:752: illegal start of type
    [javac]       HashMap<String, Symbol> addresses = new HashMap<>();
    [javac]                                                       ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:945: illegal start of type
    [javac]     ArrayList<String> output = new ArrayList<>();
    [javac]                                              ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1049: illegal start of type
    [javac]     this.moteInterfacesClasses = new ArrayList<>();
    [javac]                                                ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1077: <identifier> expected
    [javac]     } catch (NoSuchAlgorithmException | IOException e) {
    [javac]                                      ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1077: '{' expected
    [javac]     } catch (NoSuchAlgorithmException | IOException e) {
    [javac]                                        ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1077: not a statement
    [javac]     } catch (NoSuchAlgorithmException | IOException e) {
    [javac]                                                     ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1077: ';' expected
    [javac]     } catch (NoSuchAlgorithmException | IOException e) {
    [javac]                                                      ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1090: illegal start of expression
    [javac]   public static String generateUniqueMoteTypeID(MoteType[] existingTypes, Collection reservedIdentifiers) {
    [javac]   ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1090: illegal start of expression
    [javac]   public static String generateUniqueMoteTypeID(MoteType[] existingTypes, Collection reservedIdentifiers) {
    [javac]          ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1090: ';' expected
    [javac]   public static String generateUniqueMoteTypeID(MoteType[] existingTypes, Collection reservedIdentifiers) {
    [javac]                       ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1090: '.class' expected
    [javac]   public static String generateUniqueMoteTypeID(MoteType[] existingTypes, Collection reservedIdentifiers) {
    [javac]                                                            ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1090: ';' expected
    [javac]   public static String generateUniqueMoteTypeID(MoteType[] existingTypes, Collection reservedIdentifiers) {
    [javac]                                                                         ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1090: ';' expected
    [javac]   public static String generateUniqueMoteTypeID(MoteType[] existingTypes, Collection reservedIdentifiers) {
    [javac]                                                                                                         ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1140: ';' expected
    [javac]   public JComponent getTypeVisualizer() {
    [javac]                                      ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1189: ';' expected
    [javac]   public Collection<Element> getConfigXML(Simulation simulation) {
    [javac]                                          ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1189: ';' expected
    [javac]   public Collection<Element> getConfigXML(Simulation simulation) {
    [javac]                                                                ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1190: illegal start of type
    [javac]     ArrayList<Element> config = new ArrayList<>();
    [javac]                                               ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/contikimote/ContikiMoteType.java:1230: ';' expected
[... 224 lines cut ...]
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/plugins/Visualizer.java:782: illegal start of expression
    [javac]   public static void unregisterVisualizerSkin(Class<? extends VisualizerSkin> skin) {
    [javac]          ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/plugins/Visualizer.java:782: ';' expected
    [javac]   public static void unregisterVisualizerSkin(Class<? extends VisualizerSkin> skin) {
    [javac]                ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/plugins/Visualizer.java:782: illegal start of expression
    [javac]   public static void unregisterVisualizerSkin(Class<? extends VisualizerSkin> skin) {
    [javac]                                                     ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/plugins/Visualizer.java:782: illegal start of expression
    [javac]   public static void unregisterVisualizerSkin(Class<? extends VisualizerSkin> skin) {
    [javac]                                                       ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/plugins/Visualizer.java:782: ';' expected
    [javac]   public static void unregisterVisualizerSkin(Class<? extends VisualizerSkin> skin) {
    [javac]                                                                                    ^
    [javac] /home/qiuxin/contiki/tools/cooja/java/org/contikios/cooja/plugins/Visualizer.java:786: illegal start of expression
    [javac]   private void handlePopupRequest(Point point) {
    [javac]   ^
    [javac] 100 errors

BUILD FAILED
/home/qiuxin/contiki/tools/cooja/build.xml:77: Compile failed; see the compiler error output for details.

Total time: 1 second
```

cmorty, 2015-05-13:
Maybe wrong java. What does `java -version` say?

xinqiu, 2015-05-14:
Thank you for responding me.
```
java version "1.6.0_35"
OpenJDK Runtime Environment (IcedTea6 1.13.7) (6b35-1.13.7-1ubuntu0.14.04.1)
OpenJDK 64-Bit Server VM (build 23.25-b01, mixed mode)
```

cmorty, 2015-05-14:
Make sure you use java 1.7+. Either uninstall openjdk-6 or google for java-alternatives.

cmorty, 2015-05-15:
You should probably close this.

closed 2015-05-17 by xinqiu