contiki-os/contiki #1073: serialdump-linux executable is broken in instant-contiki

alignan, 2015-05-17:
From https://github.com/contiki-os/contiki/pull/1048

When cloning a vanilla contiki in Instant Contiki and executing the serialdump-linux fails to run, shows
`./serialdump-linux -h
bash: ./serialdump-linux: cannot execute binary file`

Recompiling and running again shows a different error:
`./serialdump-linux /dev/ttyUSB0
connecting to /dev/ttyUSB0 (115200)
open: Invalid argument`

However the device is connected:
`Z1RC00848  /dev/ttyUSB0     Silicon Labs Zolertia Z1`

Reverting back to the previous pre-compiled object at 09a9f0d00a1bd8065f3866c3c577f949569fdf99 works as expected.

g-oikonomou, 2015-05-17:
32- vs 64-bit thing?

alignan, 2015-05-17:
That was my first guess, but recompiling didn't work.

cmorty, 2015-05-18:
@alignan I was able to reproduce the issue on another system. Could you try https://github.com/cmorty/contiki/tree/pull/serialdump

alignan, 2015-05-18:
Sure! here are the results, the current binary:

`$ ./serialdump-linux /dev/ttyUSB1
bash: ./serialdump-linux: cannot execute binary file`

Recompiling brings joy:
`$ make serialdump
cc     serialdump.c   -o serialdump`

`

cmorty, 2015-05-18:
Probably someone should recompile the binaries. Shipping binaries isn't the best idea anyway. Or ship precompiled ones and copy them in the Makefile if there is no CC available. But that's for someone else to ponder about. :)

alignan, 2015-09-07:
Closed via https://github.com/contiki-os/contiki/pull/1076

closed 2015-09-07 by alignan

zhangtuoparis13, 2016-11-30:
hello, @alignan @g-oikonomou , i am trying to set up contiki on Raspbian within Raspberry Pi3, 
i have done all the necessary and all works fine, but finally met a problem when i make login:

```
pi@raspberrypi:~/contiki/examples/hello-world $ make login
using saved target 'zoul'
../../tools/sky/serialdump-linux -b115200 /dev/ttyUSB0
../../tools/sky/serialdump-linux: 1: ../../tools/sky/serialdump-linux: Syntax error: ")" unexpected
../../platform/zoul/Makefile.zoul:115: recipe for target 'login' failed
make: *** [login] Error 2
```
It seems to be the problem about **serialdump-linux** issue related to here.
Could you please tell me how to figure it ?

Thx a lot!

g-oikonomou, 2016-11-30:
Which probably tells us that we may want to reconsider shipping binaries... :)

zhangtuoparis13, 2016-12-01:
Yes, it works! but it should be in repo sky.
so `cd tools/sky && rm serialdump-linux && make serialdump && mv serialdump serialdump-linux `  
thx kindly @alignan @g-oikonomou  :)