contiki-os/contiki #1089: cc26xxware timer.c is not built

tonnenpinguin, 2015-05-27:
Hi guys!

When I try to build my project that utilizes the function ti_lib_timer_int_register() I get a linker error stating the following:

cc2650.c:(.text.trf_startClock+0x54): undefined reference to `NOROM_TimerIntRegister'
collect2: error: ld returned 1 exit status

Other function calls to the same library do not cause these issues. Looking through the build process I have seen that all of the other ti_lib_timer_* function calls I make are either declared inline or offer a ROM version (which is automatically chosen AFAIK, right?)

The output during the build process also suggests, that the timer.c file never gets built.

[...]
  CC        ../../../core/net/rpl/rpl-of0.c
  CC        ../../../core/net/rpl/rpl-timers.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/adi.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/aon_batmon.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/aon_event.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/aon_ioc.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/aon_rtc.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/aon_wuc.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/aux_adc.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/aux_smph.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/aux_tdc.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/aux_timer.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/aux_wuc.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/chipinfo.c
../../../cpu/cc26xx/lib/cc26xxware/driverlib/chipinfo.c: In function 'ChipInfo_GetHwRevision':
../../../cpu/cc26xx/lib/cc26xxware/driverlib/chipinfo.c:120:4: warning: enumeration value 'FAMILY_Unknown' not handled in switch [-Wswitch]
    switch( ChipInfo_GetChipFamily() ) {
    ^
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/cpu.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/crypto.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/ddi.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/debug.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/driverlib_release.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/event.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/flash.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/flashsafe.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/gpio.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/i2c.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/i2s.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/interrupt.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/ioc.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/osc.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/prcm.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/pwr_ctrl.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/rfc.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/setup.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/smph.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/ssi.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/sys_ctrl.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/systick.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/trng.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/udma.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/vims.c
  CC        ../../../cpu/cc26xx/lib/cc26xxware/driverlib/watchdog.c
  CC        ../../../core/net/linkaddr.c
  CC        ../../../core/net/nbr-table.c
[...]

As the Makefile.include uses a wildcard to include all .c files in the module I don't quite understand why this is happening. Could you please give me a pointer to where I can start to fix this issue?

Thank you very much!

g-oikonomou, 2015-05-27:
This is caused because there is a file called `timer.c` in the Contiki core too.

https://github.com/contiki-os/contiki/tree/master/platform/srf06-cc26xx#filename-conflicts-between-contiki-and-cc26xxware

tonnenpinguin, 2015-05-27:
ah, true!
Thank you so much and sorry for opening all those issues!

closed 2015-05-27 by tonnenpinguin