contiki-os/contiki #1095: CC26xx: SensorTag 2.0 CC-Devpack-Debug

schuschu, 2015-06-01:
Since the new cc26xxware and the changes how the UART is handled I am no longer able to get any serial output using the devpack (remapping the UART to IOID_29 and IOID_28).

Could you please elaborate what I need to configure/do to get them working as the board.h mentions that at least one direction should be implemented already (neither works for me though).

schuschu, 2015-06-01:
should have posted a minute later. RX and TX need to be crossed it seems (at least for my devpacks, I remember reading they reversed them in some revision)

jonnteolsson, 2015-06-02:
I can confirm that the RX and TX lines are switched. This will be addressed in a future upgrade of the DEBUG DEVPACK. For now one need:

$ git diff
diff --git a/platform/srf06-cc26xx/sensortag/board.h b/platform/srf06-cc26xx/sensortag/board.h
index fc293f5..b8e3c29 100644
--- a/platform/srf06-cc26xx/sensortag/board.h
+++ b/platform/srf06-cc26xx/sensortag/board.h
@@ -100,11 +100,11 @@
  * Those values are not meant to be modified by the user
  * @{
  */
-#define BOARD_IOID_DP4_UARTRX     IOID_28
-#define BOARD_IOID_DP5_UARTTX     IOID_29
+#define BOARD_IOID_DP4_UARTRX     IOID_29
+#define BOARD_IOID_DP5_UARTTX     IOID_28

 #define BOARD_IOID_UART_RX        BOARD_IOID_DP4_UARTRX
-#define BOARD_IOID_UART_TX        IOID_16
+#define BOARD_IOID_UART_TX        BOARD_IOID_DP5_UARTTX

 #define BOARD_IOID_UART_CTS       IOID_UNUSED
 #define BOARD_IOID_UART_RTS       IOID_UNUSED

schuschu, 2015-06-05:
I got my hands on a rev 1.2.0 devpack today and can confirm the pins are the opposite of my 1.0.0 devpack.

g-oikonomou, 2015-06-05:
Thanks for the feedback. I also received a rev 1.2.0 recently which I tried and it seemed to work fine.

We don't really want to support v1.0.0 since the number of those out there is limited and doesn't justify the effort. It could be easily documented though.

So, just to re-iterate, all you have to do with the v1.2.0 is re-map RX to `IOID_28`. Right?

schuschu, 2015-06-05:
Right, re-mapping rx is enough. You wouldn't happen to know if there is a firmware update on the way for the early adopters? (on the other hand a little define which changes it for rev 1.0.0 would also be welcome so i don't have to patch it myself whenever board.h changes)

g-oikonomou, 2015-06-05:
What firmware update are you referring to? The devpack?

schuschu, 2015-06-05:
Yes. As the devpack has a dfu mode I was kinda hoping for a fix from ti.

g-oikonomou, 2015-06-05:
This is something I have no clue about I'm afraid.

schuschu, 2015-06-05:
Ok, Thanks for the help though.

closed 2015-06-19 by g-oikonomou

karlrenner214, 2015-07-12:
I am new to using debug devpack and sensor tag.  What is the procedure for reading data from the serial port using debug devpack version 1.2 and cc2650 sensor tag running the cc2650-demo?  I have tried using hyperterminal connected to com4 but nothing is seen.

karlrenner214, 2015-07-12:
The demo is cc26xx-demo.

g-oikonomou, 2015-07-12:
Open `platform/srf06-cc26xx/sensortag/board.h` and apply the change discussed above:

```
-#define BOARD_IOID_UART_TX        IOID_16
+#define BOARD_IOID_UART_TX        BOARD_IOID_DP5_UARTTX
```

In hyperterm, you want to select 115200, 8N1, no flow control. I'm not sure which COM port you should be using.

The debugger must be attached to the tag and a coin must be inserted in the tag before plugging the debugger to your PC.

karlrenner214, 2015-07-12:
What does coin mean?  The existing battery?  The ports are XDS110 Class Application/User UART (COM4)  and XDS110 Class Auxiliary Data Port (COM3).

g-oikonomou, 2015-07-12:
Yes, the battery. It must be in before connecting the debugger. It's OK to remove afterwards. I've no idea about the COM port, I only use the debugger on OS X.

karlrenner214, 2015-07-12:
I connected hyperterminal to COM4 with the settings that you specified and it now works.  Thanks!