contiki-os/contiki #1100: CoAP block size transfer parse error

arthurfabre, 2015-06-04:
Using Contiki master on a Zolertia Z1, I'm experiencing errors with CoAP block sizes.

If a client GETs a resource (without specifying any block2 options) whose handler requests the use of blocks (by setting the offset variable), subsequent messages from the client (to request the additional blocks) will contain block2 options (such as the block_size).
The block_size option is not handled correctly, which leads to a wrong offset, which causes the wrong data to be sent.

I've verified using WireShark that the expected block_size is correctly sent by the client, but the block_size received by `er-coap-engine.c` is incorrect.

All of the code that deals with the block2 option parsing appears to be correct.
Weirdly enough, adding a `printf` statement as follows on line `644` of `er-coap.c` (`apps/er-coap/`) resolves the issue.
`printf("%d\n", coap_pkt->block2_size);`

I'm more than willing to look into this further, but I have no idea where to start - this is very weird.

A zip file containing logs obtained over serial of a working and non working sequence of messages, as well as the associated wireshark packet dumps and a patch to add the print statement in can be found [here](https://files.arthurfabre.com/?t=1e01192f02c0bfc0ce9175fb61be28ee).

arthurfabre, 2015-09-12:
This seems to have been fixed by 5403b2a211034144137ab0ec52b995a3c23d8e37, looking back on this I'm not quite sure how I came to the conclusion that `printf` fixed it, seeing as I had already fixed the default block size.

closed 2015-09-12 by arthurfabre

smnblt, 2019-06-12:
I'm still facing this issue even if I have 5403b2a. It happens in cooja only with emulated motes, thus cooja motes are fine. Strangely, the printf workaround works! If the issue happens only in cooja, I suppose that it is something related to some wrong timings!?

More info:
In my case the setup comprises a Border Router and a CoAp server, which is configured with COAP_MAX_BLOCK_SIZE = 64 and with a coap resource that produces 256B payload (thus, I expect 256/64 = 4 blocks)
With copper, I ask for 64B blocks. Then, I correctly receive the first block of 64. But, following ones will have 16B as size.