contiki-os/contiki #1106: icmpv6 message are forwarded by error

laurentderu, 2015-06-09:
The code managing the icmpv6 packets in uip6.c requires that the handler set uip_len to 0 when it does not have to send a reply packet.

```
  /*
   * Search generic input handlers.
   * The handler is in charge of setting uip_len to 0
   */
```

However, the DIO and DAO handlers simply return when an invalid message or option is received (e.g. DAO with invalid rank or from an unknown DAG). As uip_len is not set, the stack tries to send the packet again, resulting of the sending of a NS message, with target set to the host own IP, to resolve the destination address.

Either we should add a call to uip_clear_buf() as proposed in #927 or change the behavior of the imp management code and force uip_len to zero when the handler returns. If a reply must be sent, then the handler should instead call uip_icmp6_send().

g-oikonomou, 2015-06-09:
You are right, this needs fixed. I have a slight preference to `uip_clear_buf()` because i) it puts less strain on the stack and ii) it's consistent with other parts of the code, but I understand it will be needed in many places and it could be error prone. `goto discard;` could also work, as per the handlers in `uip-ndc6.c`.

Irrespective of which one we end up choosing, I think we should do it consistently across the board. i.e. either we add `_clear_buf` everywhere, or we remove `uip_len = 0` from everywhere (e.g. ND code)

adamdunkels, 2015-08-27:
This should be fixed by #1218, so I'm closing this - @laurentderu, please reopen if you think more fixing is needed!

closed 2015-08-27 by adamdunkels