contiki-os/contiki #1110: No common ISR Enable/Disable or Atomic Section Interface

bkozak-scanimetrics, 2015-06-10:
There appears to be no common interface for interrupt disable / enable or for atomic sections.

I think that this would be very useful for any platform agnostic applications which need atomic sections and crucial if ever someone needs to create an atomic section in any of the core drivers.

I've created such an interface which you can compare at:
https://github.com/contiki-os/contiki/compare/master...bkozak-scanimetrics:interrupt-api?expand=1 
at this time I only have an implementation for cc26xx.

I've also modified the cc26xx radio driver to use the interface to demonstrate its use (and help to test it).

I'd like like to get some community feedback on this before opening a pull request since I think that this would need implementations from every platform in order for this to be truly useful and since I think that this is very important to get right.

Thanks

marcas756, 2015-06-14:
:+1: 

But I would also consider restoring the previous interrupt state instead of just enabling interrupts again.
I use the following for my arch (MSP430 & ICC430)  at the moment (does currently not fit into the arch independend  design for contiki) :

```C
#define CRITICAL_SECTION_BEGIN() \
    { \
        istate_t critical_section_istate = __get_interrupt_state(); \
        __disable_interrupt();

#define CRITICAL_SECTION_END() \
        __set_interrupt_state(critical_section_istate); \
    }
```

bkozak-scanimetrics, 2015-06-15:
Yes,  interrupts_disabled does store the interrupt state. I believe that my code is equivalent to what you intend with your example.