contiki-os/contiki #1134: CC26xx missing ti_lib_int_master_enable in radio driver

bkozak-scanimetrics, 2015-06-19:
As per my discussion with @tonnenpinguin on PR #1117 we have noticed that in the ISR function cc26xx_rf_cpe1_isr in the file [cc25xx-rf.c](https://github.com/contiki-os/contiki/blob/master/cpu/cc26xx/dev/cc26xx-rf.c) interrupts are disabled and will not be re-enabled if power_up() is called and fails.

g-oikonomou, 2015-06-19:
So the problem here is that `cpe1_isr` should never enter when the RF core is off. However, I've actually seen it happen on the very odd occasion and I cannot explain under what conditions.

I think it would be OK to leave the global interrupt alone in this one. What we definitely cannot do is touch `RFC_DBELL_O_RFCPEIFG` if the RF core is not accessible, as that would throw a bus error.

bkozak-scanimetrics, 2015-06-19:
Not sure exactly what you mean with

> I think it would be OK to leave the global interrupt alone in this one.

If you mean that you don't need to disable interrupts here then I don't see why you need to either. Otherwise it would seem appropriate to me to be appropriate to print an error message and force a reset if power_up() fails (seems like an unrecoverable error to me).

> I've actually seen it happen on the very odd occasion and I cannot explain under what conditions.

I haven't seen it yet but I'll keep an eye out for it.

g-oikonomou, 2015-06-19:
> If you mean that you don't need to disable interrupts here then I don't see why you need to either.

Yes, this is what I meant.

> Otherwise it would seem appropriate to me to be appropriate to print an error message and force a reset if power_up() fails (seems like an unrecoverable error to me).

Again, agreed. If the handler enters with the RFcore off then we have a problem. If `power_up()` fails then we have an even bigger problem (unrecoverable, as you say).

g-oikonomou, 2016-05-14:
This was fixed a while ago but it was not closed.

closed 2016-05-14 by g-oikonomou