contiki-os/contiki #1136: Automatically detect addition / removal of doxy warnings

g-oikonomou, 2015-06-20:
Following the discussion in #1133, I think that having to adjust `doxyerrors.cnt` every time we fix some doxygen warnings is a little tedious and arguably unnecessary.

I've still not looked into implementing this, but here is how I think it could work in a nutshell:

* Build documentation for the pull. Store the `doxygen.log` somewhere under a different name (assume `pull.log`)
* Checkout the branch on top of which the pull is based (which may or may not be master, so we can't simply checkout master). I've not tested it, but it looks like the Travis environment variable `TRAVIS_BRANCH` is what we want here.
* Clean the documentation, build it again.
* Compare additions / removals between the two by using something like `diff doxygen.log pull.log | grep \>`

If the pull adds warnings, fail and show the warnings diff so the contributor can immediately see what has been added:

```
$ diff doxygen.log pull.log | grep \>
> [generated]:-1: warning: explicit link request to 'stStackStatusHandler()' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:943: warning: explicit link request to 'ST_GET_LINK_KEY_WHEN_JOINING' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:980: warning: explicit link request to 'ST_SECURITY_LEVEL' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:1014: warning: explicit link request to 'stSetInitialSecurityState()' could not be resolved
> [generated]:-1: warning: explicit link request to 'stStackStatusHandler()' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:943: warning: explicit link request to 'ST_GET_LINK_KEY_WHEN_JOINING' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:980: warning: explicit link request to 'ST_SECURITY_LEVEL' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:1014: warning: explicit link request to 'stSetInitialSecurityState()' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:628: warning: explicit link request to 'ST_APS_UNICAST_MESSAGE_COUNT' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:885: warning: explicit link request to 'stGetNodeId()' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:895: warning: explicit link request to 'stGetPanId()' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:921: warning: explicit link request to 'ST_REQUIRE_ENCRYPTED_KEY' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:364: warning: explicit link request to 'ERASE_CRITICAL_THRESHOLD' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:366: warning: explicit link request to 'halSimEepromErasePage()' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:379: warning: explicit link request to 'ERASE_CRITICAL_THRESHOLD' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:382: warning: explicit link request to 'halSimEepromErasePage()' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:395: warning: explicit link request to 'SIM_EEPROM_ERASE_PAGE_RED' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:397: warning: explicit link request to 'halSimEepromErasePage()' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:442: warning: explicit link request to 'TOKEN_MFG_NVDATA_VERSION' could not be resolved
> /Users/cexgo/Documents/workspace/contiki/cpu/stm32w108/hal/error-def.h:443: warning: explicit link request to 'TOKEN_STACK_NVDATA_VERSION' could not be resolved
```

If the pull does not add warnings, then the output of this command will be empty:

```
$ diff doxygen.log pull.log | grep \>
$
```

Lastly we can print out the current number of warnings for informative purposes if we want to.

Any thoughts?

bthebaudeau, 2015-06-20:
Looks good.

> Checkout the branch on top of which the pull is based (which may or may not be master, so we can't simply checkout master). I've not tested it, but it looks like the Travis environment variable `TRAVIS_BRANCH` is what we want here.

http://docs.travis-ci.com/user/environment-variables/#Default-Environment-Variables
`TRAVIS_BRANCH:For builds not triggered by a pull request this is the name of the branch currently being built; whereas for builds triggered by a pull request this is the name of the branch targeted by the pull request (in many cases this will be master).`

`TRAVIS_BRANCH` is indeed what we want here. But it's not simply this branch that should be checked out to run doxygen on it. It's rather the best common ancestor between this branch and the PR's, which can be known using `git merge-base`. This is because the target branch may have changed in the meantime.

g-oikonomou, 2015-06-20:
> This is because the target branch may have changed in the meantime.

Good point, you mean if e.g. a different pull gets merged in the meantime. Right?

Further, I think we can run the comparison only if `TRAVIS_PULL_REQUEST` is not `false`, automatically succeed otherwise.

bthebaudeau, 2015-06-20:
> Good point, you mean if e.g. a different pull gets merged in the meantime. Right?

Exactly, e.g.:
 - PR created, not changing anything for doxygen,
 - doxygen warnings fixed in the target branch thanks to new commits independent of the PR above,
 - new commits added to the PR, not changing anything for doxygen,
 - Travis runs again,
 - if run against the target branch rather than the best common ancestor between this branch and the PR's, then the comparison wrongly finds that the PR adds warnings.

> Further, I think we can run the comparison only if `TRAVIS_PULL_REQUEST` is not `false`, automatically succeed otherwise.

Probably. In what circumstances may Travis be run without a PR?

g-oikonomou, 2015-06-20:
> Probably. In what circumstances may Travis be run without a PR?

Travis fires for a branch every time one or more commits get pushed to said branch (including if it's for a new branch)

bthebaudeau, 2015-06-20:
> Travis fires for a branch every time one or more commits get pushed to said branch (including if it's for a new branch)

Then the documentation of `TRAVIS_BRANCH` becomes ambiguous: if commits are added later to a branch previously set for a PR, does it consider the build to have been triggered by the PR or only by the new commits? Technically, according to the definition, this should be the new commits, but this is not what we need here. Testing would tell.

For new branches unrelated to PRs, the best common ancestor with master could be used as a reference, but maybe this does not always make sense, depending on the purpose of the branch.

g-oikonomou, 2015-06-20:
Assume I create a branch `pull/fix-something` and I open a pull with it against `contiki-os/contiki/master`. Travis will run two jobs:

* One under `g-oikonomou/contiki` for branch `pull/fix-something`. In this case, `TRAVIS_BRANCH` will be a reference to `pull/fix-something` and `TRAVIS_PULL_REQUEST` will be `false`
* A second one under `contiki-os/contiki` for the merge. `TRAVIS_BRANCH` will be `master` on `contiki-os/contiki` and  `TRAVIS_PULL_REQUEST` will be some sort of reference to the pull (e.g. `1976`).

If additional commits are added to the branch (and therefore the pull), the two jobs will be triggered again with the same values for the env. vars.

I'm not sure whether those variables will be `pull/fix-something` or `g-oikonomou/contiki/pull/fix-something` or what have you.

bthebaudeau, 2015-06-20:
Then it seems OK. Displaying this variable in a test would give its exact format.

`TRAVIS_COMMIT_RANGE` could also be used, with the first parent of the first commit (if this is not a merge commit).

tsparber, 2016-01-25:
Obsolete since #1344 is merged :tada:

closed 2016-01-25 by g-oikonomou