contiki-os/contiki #1142: CSMA backoff logic and re-TX count ignore MAC_TX_COLLISION

g-oikonomou, 2015-06-24:
In `csma.c :: packet_sent()` ([L286](https://github.com/contiki-os/contiki/blob/master/core/net/mac/csma.c#L286)), we start with this:

```
switch(status) {
case MAC_TX_OK:
case MAC_TX_NOACK:
  n->transmissions += num_transmissions;
  break;
case MAC_TX_COLLISION:
  n->collisions += num_transmissions;
  break;
...
```
Thus, if the RDC signals `MAC_TX_COLLISION` we increment `n->collisions`, but _not_ `n->transmissions`

This results in two problems:

* The "abort if re-TX count exceeded" logic fails
```
if(n->transmissions < metadata->max_transmissions) {
  /* Do stuff */
} else {
  /* Drop the frame */
}
```
This logic ignores the number of times the RDC driver returned `MAC_TX_COLLISION`. This check will keep failing and CSMA will keep retrying until `MAC_TX_OK` or until we receive `MAC_TX_NOACK` for a number of times exceeding `metadata->max_transmissions`.

* The backoff logic does not seem to take into account `n->collisions` either:

```
num_tx = n->transmissions;
...
backoff_exponent = num_tx;
```

IMHO when RDC returns `MAC_TX_COLLISION`, the failed TX attempt should be taken into account for both parts of the CMSA logic

simonduq, 2015-06-24:
I agree that collisions need to be accounted for. My opinion is we should aim for making Csma (+Nullrdc) compliant with IEEE802.15.4 nonbeacon-enabled mode. That would mean:
* 1) perform CCA for a maximum of `macMaxCSMABackoffs` (default: 4), with exponential back-off between attempts;
* 2) transmit immediately after a successful CCA. In case of failure, iterate (reset backoff exponent, goto 1) until exceeding `macMaxFrameRetries` (default: 3).

laurentderu, 2015-09-04:
Is there someone working of this issue ? If not, I would propose a quick fix to avoid infinite retransmissions because of collisions.

alignan, 2016-03-22:
I have stumbled on to this as well, if there is a  constant channel interference it loops forever doing CCA checks and retransmission attempts.

A dirty fix is to check `n->collisions` with `metadata->max_transmissions` also, but having a separate `max_collisions` it would be better.

bkozak-scanimetrics, 2016-03-22:
I've never agreed with the approach of having a separate ```max_transmissions``` value per transfer. Even if you need this value to be run-time configurable (although I don't really see why) there should be a single value for all transmissions.

So if a separate value for maximum collisions is used (and I would agree that this is better) I think it should just be a constant.

atiselsts, 2016-03-23:
I have seen this separate handling of transmissions and collisions creating
problems as well. In particular, the channel blacklisting code for
multichannel ContikiMAC (http://user.it.uu.se/~atiel485/elsts_ewsn2016.pdf)
initially did not work in a testbed, because it only looked at transmission
failures, therefore failed to blacklist channels on which transmissions
were never started.

I fixed that in my repository by changing some MAC protocol code, but if
such a blacklisting was done at the application level, it would be
impossible to account for collisions at all. (Only the number of
transmissions is reported back by Contiki LLSEC/MAC layers.) So this is a
kind of limitation for building more advanced stuff on top of Contiki stack.

On Tue, Mar 22, 2016 at 4:27 PM, Billy Kozak <notifications@github.com>
wrote:

> I've never agreed with the approach of having a separate max_transmissions
> value per transfer. Even if you need this value to be run-time configurable
> (although I don't really see why) there should be a single value for all
> transmissions.
>
Surely there can be different QoS classes for traffic, each of them
requiring a different maximal number of retries.

bkozak-scanimetrics, 2016-03-24:
> Surely there can be different QoS classes for traffic, each of them requiring a different maximal number of retries.

I would think that this should be handled at a higher layer. The MAC layer should only be responsible for ensuring that channel access is done correctly which doesn't have anything to do with whether the application writer really wants the packet to be received no matter what.

alignan, 2016-04-28:
Closed https://github.com/contiki-os/contiki/pull/1579

PR #1579 (2016-04-28): Use collisions for csma backoff and retry
This PR fixes #1142 

The choice of default value for ```CSMA_MAX_MAX_COLLISIONS``` is working for me but is otherwise arbitrary. Please speak up if you know a better value for this constant.

I have some other concerns about the effectiveness of the exponential back-off constants now that collisions are accounted for (and see also my issue at #1571) but I think that this should be good enough for now.
closed 2016-04-28 by alignan