contiki-os/contiki #1147: AES-128 timing vulnerability

akdjka, 2015-06-30:
AES128 implementation does not run in constant time, the culprit being a branch in galois_mul2().
I suggest to change galois_mul2 to the following.

static uint8_t
galois_mul2(uint8_t value)
{
unsigned xor_val = (value>>7) * 0x1b;
	return ((value << 1) ^ xor_val);
}

As a side effect, on Cortex M0+ the code is both smaller and faster by a tiny amount.

cmorty, 2015-07-04:
Please make a pull-request with your chages. There are instructions on the wiki how to do so.

akdjka, 2015-07-09:
Forking your project is much further off my way than copying and pasting a piece of code is for a contiki developer and I don't intend to do it.

simonduq, 2015-08-03:
I created a PR https://github.com/contiki-os/contiki/pull/1183

simonduq, 2015-08-31:
now merged!

closed 2015-08-31 by simonduq