contiki-os/contiki #1148: uIP6 code-base - 255 byte stack compliance?

joakimeriksson, 2015-06-30:
Hello,

Do we strictly need to support device with 255 bytes stack limitation for the full 6LoWPAN stack in Contiki OS anymore? Lots of lots of the code is crippled with module wide variables that would be better to allocate on the stack the few times they are actually used (just temporary variables).

This is how it looks in many places:
---
static uint8_t *nd6_opt_llao;   /**  Pointer to llao option in uip_buf */
static uip_ds6_nbr_t *nbr; /**  Pointer to a nbr cache entry*/
static uip_ds6_defrt_t *defrt; /**  Pointer to a router list entry */
static uip_ds6_addr_t *addr; /**  Pointer to an interface address */
---

Are we expecting 8051, 6502, etc. to be mainstream IoT devices or can we clean-up the code?
(I have not yet taken a look at the actual difference if we would go ahead an allocate the variables when needed - but the case is that now all variables are always allocated but changing it would make them only be allocated when needed - and typically there is no functions using all of them at the same time which means that any device with >255 bytes on the stack will handle this without problems).

oliverschmidt, 2015-06-30:
Hi Joakim,

> Do we strictly need to support device with 255 bytes stack limitation for
> the full 6LoWPAN stack in Contiki OS anymore?
>
> I wasn't aware that Contiki supported those devices at all...

> Are we expecting 8051, 6502, etc. to be mainstream IoT devices or can we
> clean-up the code?
>
I can only talk about the 6502...

With the toolchain used for Contiki (cc65) there's no limitation 255 bytes
stack limitation. And to be honest I don't believe that any average Contiki
program can live inside 255 bytes of stack...

> ... and typically there is no functions using all of them at the same time
> ...
>
> Yeah, that's the general idea of a stack ;-)

Afaik all those static vars were introduced because the code generated by
cc65 for accessing them is (much) smaller/faster than for automatic vars
(aka vars on the stack).

Regards,
Oliver

>

joakimeriksson, 2015-06-30:
So then I would say there is hope - I guess the same things go for the toolchain of 8051? Or do should we expect that compiler to have issues?

g-oikonomou, 2015-07-01:
> Do we strictly need to support device with 255 bytes stack limitation for the full 6LoWPAN stack in Contiki OS anymore?

> Are we expecting 8051, 6502, etc. to be mainstream IoT devices or can we clean-up the code?

The last "stable" version for CC2430 / CC2530 (8051 / SDCC) is from Jan 09 and I have pretty-much given up on them in terms of running 6LoWPAN / RPL.

I am going to get rid of the Sensinode platform in due course.

The cc2530dk platform is still very useful for some purposes (the CC2531 USB dongle being very very good as a sniffer and as a slip-radio). Thus, whatever cleanup you do, please make sure cc2530 examples keep compiling. Notably, make sure to not intermingle variable declarations with code (which is something you shouldn't be doing anyway), because SDCC will fail with an obscure error.

oliverschmidt, 2015-07-01:
Hi,

> Notably, make sure to not intermingle variable declarations with code
> (which is something you shouldn't be doing anyway), because SDCC will fail
> with an obscure error.
>
I presume you're talking about variable definitions here... This is - as
you say - to be avoided cause it isn't allowed in C89. And cc65 doesn't
support it either.

Regards,
Oliver

>