contiki-os/contiki #1154: Debugging TI sensortag web-demo with SRF06 gives error: could not open flash to load config

weijian19391, 2015-07-08:
Hi 

Does anyone knows how to resolve that error? Also, when starting the Sensortag, it shows 
````
Starting Contiki-2.6-2409-ge7aba9b
````
I was under the impression that we're at contiki 2.7, is there some update that i need to perform? 
Thank you

jonnteolsson, 2015-07-08:
Hi, The SRF06 does not include an external flash to save config to. It is only valid for the CC2650 SensorTag. 

Regarding the startup printf:
If you are not building from within a git dir it falls back to a header (which is your case here), so you are 2409 commits ahead of the latest tag, which is 2.6. There is no 2.7 tag

If you build for within a git dir you will get Contiki 3.x and then the git commit id g"xxxxxxxx"

weijian19391, 2015-07-09:
@jonnteolsson Thanks for your reply! 

Do you mean that when i am debugging, i'm actually programming the SRF06 instead of the CC2650STK? Because to the best of my knowledge, the error appeared from one of the debug printf that arise from the code running on the sensortag, not the srf06, so how is it that the debug will be look for an external flash on the srf06 instead of that from the sensortag?

g-oikonomou, 2015-07-09:
As Jonas says, the startup output is perfectly normal.

In terms of the question in the subject: How are things connected to each other? Tag to the Srf with a 10-pin JTAG cable? Something else? If so, how are you getting debug output from the tag?

weijian19391, 2015-07-09:
Morning @jonnteolsson (at least from where i am)!

Yes, i connected the sensortag to the srf via a 10-pin JTAG cable. 

In addition, i modified the jumper settings for XDS100v3 BYPASS(P408) according to a post on the TI forum  (https://e2e.ti.com/support/wireless_connectivity/f/158/p/411992/1483824#1483824). To summarise the post : "we use two spare JTAG signals to connect to the UART on the SRF, and we connect pin 6 to pin 11, pin 8 to pin 13 of P408"

Lastly, since i am using the SRF as the debugger instead of CC dev-pack debugger, i followed this post (https://e2e.ti.com/support/wireless_connectivity/f/158/p/419000/1493645#1493645) to modify the code from ````/platform/srf06-cc26xx/sensortag/board.h```` , editing the below two lines: 
````
#define BOARD_IOID_UART_RX IOID_17
#define BOARD_IOID_UART_TX IOID_16
````

In terms of seeing the debug, i'm using putty to open the COM port, listening on 115200N1.
Sorry for the complicated setup!

g-oikonomou, 2015-07-09:
It's not a complicated setup, I just wanted to make sure your setup was what I was suspecting.

So the problem is that the SPI CLK pin is also on `IOID_17`, same as `UART_RX` in your setup. Mapping UART_RX and SPI CLK to the same pin basically messes up external flash operation.

So you have two options:

* Keep calm and carry on without the ext flash
* Disable UART RX so that `IOID_17` can be driven by SPI. Basically by reverting `BOARD_IOID_UART_RX` to its previous value.

It is possible to do the SmartRF cable bypass for the TX pin only: just remove the jumper cable that connects `IOID_17` to TDO and leave the other one on. In doing so, you will get back ext flash operation, but you will naturally lose the ability to send chars to the UART.

A secondary issue you may notice with the ext flash is that it may throw the same error immediately after programming the tag with a new image. This will then disappear after you have fully power-cycled the tag by using the Srf power switch. A simple reset won't do it if memory serves.

(btw, you're talking to a different person :))

weijian19391, 2015-07-09:
@g-oikonomou  Opps! edited that post. hahaha. 

I see! Thanks for the clear explanation! I think i will go ahead with option one of keeping calm and carrying on as i really need the UART communication to debug the Sensortag, who's MQTT application is not working as desired. 

Once again, Thanks!

closed 2015-07-09 by g-oikonomou