contiki-os/contiki #1159: ContikiMAC and NETSTACK_RADIO.receiving_packet()

g-oikonomou, 2015-07-11:
The NETSTACK_RADIO API is unclear for `receiving_packet()` in terms of who's supposed to make sure the radio is on:

* Is it the caller's responsibility to make sure to call `NETSTACK_RADIO.on()` before calling `NETSTACK_RADIO.receiving_packet()`?

If so, then ContikiMAC is bugged [near line 581](https://github.com/contiki-os/contiki/blob/master/core/net/mac/contikimac/contikimac.c#L581), with this part of the code executed while the radio is potentially off:
```
  if(NETSTACK_RADIO.receiving_packet() || NETSTACK_RADIO.pending_packet()) {
```

* Or is it the radio's own responsibility to turn itself on (and therefore back off) within this function itself?

In which case I think we have a poor design decision leading to potentially unwanted on/off cycling. This is especially undesirable when powercycling the radio is slow if e.g. it involves firing up a XOSC.

Note that the same does not apply to `NETSTACK_RADIO.channel_clear()`. All calls in ContikiMAC are preceded by a call to `.on()`.

I am leaning towards flagging this as a bug in ContikiMAC, but I'd like to hear people's thoughts first in case I have missed something.

adamdunkels, 2015-07-11:
If the radio is off, it will not be receiving a packet. So `receiving_packet()` should return 0 (false) if the radio is off.

g-oikonomou, 2015-07-11:
Clear, thanks!

closed 2015-07-11 by g-oikonomou