contiki-os/contiki #1191: CC2650 - Several Race Conditions in Radio Driver

bkozak-scanimetrics, 2015-08-12:
The CC2650 radio driver has multiple potential race conditions associated with the ```cmd_immediate_buf```. This variable is used both from interrupt context (see ```powercycle()``` in contikimac.c) and from the main thread without first protecting by masking interrupts. Most public  radio driver functions will result in an unprotected use of cmd_immediate_buf.

g-oikonomou, 2015-08-12:
Ugh you're right... That's pretty ugly.

I think we can double-buffer with `_alloc()`, `_dealloc()`. Can you think of a scenario that would need more than 2?

bkozak-scanimetrics, 2015-08-12:
> Can you think of a scenario that would need more than 2?

I'm pretty sure 2 is fine. Should just be the RTC ISR and main thread which call radio functions.

g-oikonomou, 2015-08-12:
OK don't bother with it. I've got a big one coming so I'll make the fix part of it.

g-oikonomou, 2015-08-12:
On second thoughts, we can just allocate space for Radio Ops on stack and get it over with...

closed 2015-08-25 by g-oikonomou