contiki-os/contiki #1200: cc26xx: ROM_FlashProgram not defined

tonnenpinguin, 2015-08-18:
Hi guys!

When trying to build a program that uses cc26xxware's ROM_FlashProgram I first get a "implicitly defined function" error while building and a linker error after that.
Having looked through the code it seems like ROM_FlashProgram is not defined at all. There is only the NOROM version in flash.c that I could find.

So my question is the following:
Is that a bug in cc26xxware (Did they forget to enter the ROM calls?),
or is it a bug in contiki (Should we change the define in ti-libs.h to the NOROM version?),
or did I misinterpret something? 

The same thing is true for ROM_FlashSectorErase and ROM_FlashProgramNowait as well.

Thanks in advance!
tonnenpinguin

g-oikonomou, 2015-08-18:
You should be using `FlashProgram` (prototype  in `flash.h`). Check `rom.h` to see what ROM functions are available to you. I cannot see a `ROM_FlashProgram`, ergo that macro in `ti-glue.h` shouldn't be there in the first place.

g-oikonomou, 2015-08-18:
Bear in mind that not all xxware functions are mapped in our `ti-glue.h`. For example, there is no mapping for `FlashProgram()` or any of the functions in CC26xxware's `flash.h`. You can add them yourself in your local copy, or you can simply include `flash.h` and use the CamelCase function directly.

tonnenpinguin, 2015-08-19:
Thank you for the quick reply!

I'll directly include flash.h then!

closed 2015-08-19 by tonnenpinguin