contiki-os/contiki #1208: OWNERS file

jeez, 2015-08-20:
As described by https://github.com/contiki-os/contiki/wiki/Pull-Request-Policy , Contiki has a "merge team" which is split into smaller teams per module.

This model is similar to the OWNERS files that other projects use. Google Chromium, for instance, do that on a per directory basis: https://www.chromium.org/developers/owners-files .

The main benefit of such files is that whenever you make a change somewhere, you know upfront who you can contact to review and discuss the patch. The main drawback is that these files have to be kept up-to-date.

After #1201 is merged, what would you think about using this model on Contiki? Perhaps if having multiple files is an unnecessary overhead, we could consider to keep a single OWNERS file on the root folder centralizing the list of owners for each "module". With that the "Pull-Request-Policy" can be fully removed.

simonduq, 2015-08-26:
The benefit is unclear to me: in https://github.com/contiki-os/contiki/pull/1201 the new `CONTRIBUTING.md` includes a link to https://github.com/orgs/contiki-os/people, which is guaranteed to be up-to-date.

jeez, 2015-08-26:
[**@simonduq**](https://github.com/simonduq) commented on [Aug 26, 2015, 5:53 AM GMT-3](https://github.com/contiki-os/contiki/issues/1208#issuecomment-134902277 "2015-08-26T08:53:15Z - Replied by Github Comment Enhancer"):
> The benefit is unclear to me: in [#1201](https://github.com/contiki-os/contiki/pull/1201 "New PR merging policy") the new `CONTRIBUTING.md` includes a link to [https://github.com/orgs/contiki-os/people](https://github.com/orgs/contiki-os/people), which is guaranteed to be up-to-date.

On #1201 you suggest that https://github.com/contiki-os/contiki/wiki/Pull-Request-Policy should be deleted. The main benefit of the OWNERS file would be allowing contributors to know upfront who are the "sub-merge teams" for a specific module/directory. And keeping it up-to-date, then, would become part of the naming / un-naming people for the merge team.

The https://github.com/orgs/contiki-os/people page only lists the merge team members, but there is 0 information about who "owns" what.

simonduq, 2015-08-26:
Sorry I read "single file on the root folder" and didn't notice you proposed per-module owners. Sounds like a good idea to me then -- except we don't really have a clear list of module owners yet ;) (the list in the wiki being not really up to date)

jeez, 2015-08-26:
Oh, I thought the list was correct :/. That would be a hard-dependency for this idea to move on, I'm afraid.

simonduq, 2015-08-26:
Yes but definitely something to keep in mind if/when we get an updated list :)

mcr, 2015-08-27:
Just a comment on the pointer to contiki-os/people: as a BSD licensed code base, Contiki can enter organizations (and countries) that have very restrictive outside access policies. (I've seen this repeatedly in China).  Those entities will at some point think they invented Contiki, and may even reach out in a litigious way; more evidence in the code base (which they are too negligent to either read or remove) of who is responsible for what is useful later on.

simonduq, 2015-08-27:
Interesting thought. Let me reopen.
But do you think the author names in each .c and .h files are not enough?

alignan, 2015-08-27:
Agree with @mcr (was to reopen the issue myself).