contiki-os/contiki #1220: Power consumption issue on CC2650STK very sleepy demo

chenek, 2015-08-28:
I have test power consumption on CC2650STK very sleepy demo. I still see about 2 mA@3V in sleep mode. I think this is much higher than expectation. I also find another problem is that the current would lock at about 12 mA@3V after CC2650STK very sleepy demo runs a while and it never goes back to 2mA. Does anyone have test this and get some good results?

g-oikonomou, 2015-08-28:
Yes those numbers are too high. What kit did you use to measure?

chenek, 2015-08-28:
I use CC2650 SensorTag which is powered by a power supply. I see those current consumptions on my power supply.

chenek, 2015-08-29:
Hi George,
Do you have power consumption data when you test very sleepy demo on CC2650STK?

g-oikonomou, 2015-08-31:
I'm afraid I do not. I soon will though

chenek, 2015-09-07:
I find the while loop never runs again sometime after "etimer_set(&et_periodic, PERIODIC_INTERVAL);" in very-sleepy-demo.c. It seems etimer failed in this case.

chenek, 2015-09-08:
I try to set "#define DEBUG 1" in \contiki\core\sys\process.c to check what happens. I find it would stop at "PRINTF("process: calling process '%s' with event %d\n", PROCESS_NAME_STRING(p), ev);" after CC2650STK runs a while. The followings are the logs.

...
process: calling process 'Event timer' with event 130
process: calling process 'Event timer' with event 130
process: calling process 'Event timer' with event 130
process: calling process 'Event timer' with event 130
process: calling process 'Event timer' with event 130
process_post: Process 'Event timer' posts event 136 to process 'Ctimer process', nevents 0
process_post: Process 'Event timer' posts event 136 to process 'TCP/IP stack', nevents 1
process: calling process 'Ctimer process' with event 136
process: calling process 'Event timer' with event 130
process: calling process 'TCP/IP stack' with event 136
process: calling process 'Event timer' with event 130
process: calling process 'Event timer' with event 130
process: calling process 'Event timer' with event 130
process: calling process 'Event timer' with event 130
process: calling process 'Event timer' with event 130
process: calling process 'Event timer' with event 130
process: calling process 'Event timer' with event 130
process: calling process 'Event timer' with event 130
process_post: Process 'Event timer' posts event 136 to process 'CC13xx/CC26xx very sleepy process', nevents 0
process: calling process 'CC13xx/CC26xx very sleepy process' with event 136
process: calling process 'Event timer' with event 130
process: calling process 'Event timer' with event 130

stops and never show next.
Does anyone have any suggest so I can keep debugging?

g-oikonomou, 2015-09-10:
I had a look at power consumption the other day, but then holidays happened. I'm back looking at it and I hope to be able to report back with specifics over the next few days.

However, from what I remember, here are some thoughts:

* If memory serves, very sleepy demo was setting at ~200uA. This is way too high compared to what was intended, but the number suggests that the chip does drop to deep sleep. As discussed with Billy in various threads, to reduce this number I'd start by looking at the state of pins under deep sleep.
* When we see numbers in the region of a few mAs, this is an indication that the chip fails to deep sleep in the first place.
* There is a question mark around the web demo's net-uart functionality. When enabled, the board draws somewhere in the region of mAs from what I've seen. The net-uart basically keeps the UART clocked under sleep/deep sleep and also keeps the serial PD powered. Nevertheless, I am being told that the observed number is way too high even with those things powered up, so perhaps something prevents deep sleep altogether. Additionally, the state of TX, CTS and RTS under deep sleep should also be looked into. This is under investigation.
* Mike: Make sure you have deep sleep enabled. Simple sleep won't give you anywhere near the desired consumption numbers.

gillesDD, 2015-09-10:
Hi George,
You said "If memory serves, very sleepy demo was setting at ~200uA. This is
way too high compared to what was intended"
would you share the value that is intended, this to have an idea of the
"reference" value ?
Thanks
Gilles

bkozak-scanimetrics, 2015-09-10:
Hi @gillesDD ,
As Geroge mentioned I've also been working on power consumption on the CC2650 (but not sensortag). To help me to debug a separate issue I'm having, I've created a [simple test app for Contiki](https://github.com/bkozak-scanimetrics/CC2650_lpm_test) which I think you may find helpful.

What this test does is to, after a few seconds, disable the RTC and put the board into the lowest energy mode (where it should stay unless woken by GPIO interrupt). This will make it easy to see what the actual deep sleep mode power consumption is. Note also that the macro ```DISABLE_PINS``` can be used to disable all GPIO prior to sleeping so you can see if any GPIO are causing your power problem.

According to Table 6-5 in the [CC2650 TRM](http://www.ti.com/lit/pdf/swcu117) the sleep mode power consumption (**Standby**) should ideally be ~1μA.

g-oikonomou, 2015-09-10:
Billy this seems very handy. One comment though: the pin state selected by `lpm_pin_set_default_state()` is not necessarily the least leaky one. The best state to select also depends what each pin is connected to on the board.

bkozak-scanimetrics, 2015-09-10:
A fair point. For sensortag you might need to adjust ```disable_pins()``` if you choose to use it. I'm really not at all familiar with the peripherals on this board so I wouldn't know how best to do that atm.

chenek, 2015-11-14:
This is awesome. I will verify this as soon as I have my power measure instrument nearby.

closed 2016-01-09 by g-oikonomou

saurabhbansal90, 2017-05-09:
I also feel, for the low power operation for Contiki OS, controller should stay in sleep mode. But I have same problem, it is woken up by etimer events, as soon as it enters in deep sleep mode.

And disabling RTC will also not be possible because if RTC is disabled, then periodic tasks like checking radio RX for any incoming message, sending data on some periodic interval will not be possible.
I need to change the interval for various events using etimer and rtimer. Can anybody tell how to know the processes running various timer instances.
Another possibility is registering each module as low-power module. And I have no particular idea if TCPIP module or radio codes can be registered as low power module.