contiki-os/contiki #1229: CC2650 - Simple RPL Network Fails to Build

bkozak-scanimetrics, 2015-09-01:
I have a simple RPL network made up of two CC2650 (srf06 platform) nodes; one is a border router and the other a custom application that does nothing until a button is pressed. Up until recently this was working fine. After the major changes that took place last week however (such as #1212) the nodes are not even able to form an RPL network.

On my sniffer, I can see that RPL control packets are sent by both nodes but that no network is formed (the border router's web-page confirms this).

I have found, however, that everything mysteriously starts to work if I enable ```DEBUG``` in contikimac.c. This leads me to believe that this is some sort of a timing or LPM related issue as the debug prints from contikimac will act to slow down certain sections of code and delay sleeping.

bkozak-scanimetrics, 2015-09-01:
A brief update:

It would seem that a network will form eventually if the nodes are left long enough but it is *extremely* slow. 

After the network forms, over the air communication is very unreliable (~90% failure to reply to pings sent through the border router).

g-oikonomou, 2015-09-01:
Smells like timing of ACKs, but I have not observed anything like that. I must say my testing was mostly with slip radio + 6lbr

bkozak-scanimetrics, 2015-09-01:
> Smells like timing of ACKs

With correctly functioning devices the nodes will first try to establish an RPL network using broadcast packets. A unicast packet is only sent just before the network forms.

What I'm seeing now is that no unicast packet is sent at all (for a long time) so there is no chance for there to be a problem with the ACKs.

g-oikonomou, 2015-09-01:
Ok so the mesh node doesn't pick up the DIOs sent by the BR: either it does not wake up to detect it, or the detection fails (CCA settings?)

bkozak-scanimetrics, 2015-09-01:
> Ok so the mesh node doesn't pick up the DIOs sent by the BR

Yes, this seems to be what's happening.

> either it does not wake up to detect it, or the detection fails (CCA settings?)

Something like that seems probable to me.

tonnenpinguin, 2015-09-01:
+1 on that issue.
I had massive problems on my custom platform based on the cc2650 after merging in the changes from #1212. I did not get the chance to do any thorough tests until now though.

adamdunkels, 2015-09-01:
We found (and fixed) a number of issues with the radio driver and the clock code in #1212, which may very well cause problems like these. Fixes are coming!

bkozak-scanimetrics, 2015-09-01:
@g-oikonomou 

Looks like this is a packet spacing issue again. It will go away if we comment out the ```lpm_sleep()``` in ```transmit()``` of ieee-mode.c.

You will also need to declare the ```rfc_CMD_IEEE_TX_t cmd;``` as volatile (it should probably be volatile anyways btw).

adamdunkels, 2015-09-02:
Here are some the fixes promised earlier: #1233

The issue with `lpm_sleep()` within the radio tx function is not fixed in that pull request though.

g-oikonomou, 2015-09-02:
Billy, how about we changed `_cmd_done_en()` to accept the interrupt number as its single `uint32_t` argument and pass `IRQ_LAST_FG_COMMAND_DONE` in the IEEE driver, `IRQ_LAST_COMMAND_DONE` in the prop driver?

bkozak-scanimetrics, 2015-09-02:
Maybe just a bool instead of the uint32_t? Passing the interrupts number seems to defeat the purpose of the function IMO.

Something like:
```C
void
rf_core_cmd_done_en(bool fg)
{
  uint32_t irq = fg ? IRQ_LAST_FG_COMMAND_DONE : IRQ_LAST_COMMAND_DONE;
  HWREG(RFC_DBELL_NONBUF_BASE + RFC_DBELL_O_RFCPEIEN) = irq | ENABLED_IRQS;
}
```

Also, now that I look at this function again, pretty sure that ```HWREG(RFC_DBELL_NONBUF_BASE + RFC_DBELL_O_RFCPEIFG) = ENABLED_IRQS``` does absolutely nothing.

From the TRM *Table 23-139* (on RFCPEIFG Register):
> ... Write zero to clear flag. Write to one has no effect.

bkozak-scanimetrics, 2015-09-02:
Ah wait ...
Just got the purpose of the RFCPEIFG write.

g-oikonomou, 2015-09-02:
Not opposed to a `bool` argument as long as there is a line or two in the doxygen comment to explain how it works.

g-oikonomou, 2015-09-03:
> You will also need to declare the `rfc_CMD_IEEE_TX_t cmd;` as volatile (it should probably be volatile anyways btw).

Yes, that's a very good point

g-oikonomou, 2015-09-04:
#1231 merged - this should be fixed now. Please confirm so we can close.

bkozak-scanimetrics, 2015-09-04:
Problem's gone for me.

closed 2015-09-04 by g-oikonomou