contiki-os/contiki #1234: Run cc26xx-web-demo on CC1350 SensorTag

chenek, 2015-09-03:
I have tried to build  cc26xx-web-demo.bin for CC1350 SensorTag. I can see it is listed on 6lbr Sensors page. However, I cannot open its web page. I have no problem to open the web page using SmartRF06EB+CC1350EM with the same cc26xx-web-demo.bin. Do I have to use different BOARD define?

jonnteolsson, 2015-09-03:
Do you have the patches needed for this?

gillesDD, 2015-09-03:
Hi,
I am willing to do so but not working
did you compile the demo  using "make TARGET=srf06-cc26xx
BOARD=srf06/cc13xx"

thanks
Gilles

chenek, 2015-09-03:
I compile cc26xx-web-demo.bin using "make TARGET=srf06-cc26xx BOARD=srf06/cc13xx cc26xx-web-demo.bin". Is this correct?

gillesDD, 2015-09-03:
Well, all I can say is I just follow the Ti wiki instruction, so yes I use
the same make command.
=> you say : you board is waking up , can you see some output on serial
console  ?
did you manage with 6lbr : are you using Beagle Board plugged to the
smartf06 , on my side, smart06 does not power on when plugged to BB...
cheers
Gilles

chenek, 2015-09-03:
Do you test this with SmartRF06EB+CC1350EM or CC1350 SensorTag? My problem happens on CC1350 SensorTag.

gillesDD, 2015-09-03:
I use CC1350EM plugged to smartf06
Strange, I don't thing CC1350 comes with sensorTag, sensortag is CC2650 ,
isn't it ?
can you tell me about your SLIP radio + 6LBR HW configuration , thanks

chenek, 2015-09-03:
The Slip Radio is SmartRF06EB+CC1350EM connect to Raspberry Pi using FT232 USB-to-Serial dongle. I test on two devices. One is SmartRF06EB+CC1350EM and runs cc26xx-web-demo.bin which works fine to work with Raspberry Pi and Slip radio. Another is CC1350 SensorTag and also runs  cc26xx-web-demo.bin. But this one has problem. I can see this CC1350 SensorTag shows up on Sensors Page of 6lbr web page but I cannot access to CC1350 SensorTag's web page.

gillesDD, 2015-09-03:
are you using contiki 3.0 bleeding edge repo ?, because on this , my board
does not starts well.
can you send me the web page link to find CC1350 SensorTag ?

chenek, 2015-09-03:
What do you mean web page link to find CC1350 SensorTag?

gillesDD, 2015-09-03:
I meant, did you build this one on your self ? because TI only provide with
cc2560 sensorTag :)

chenek, 2015-09-03:
No, I get it from TI.

tonnenpinguin, 2015-09-03:
I think what @gillesDD is trying to tell you is that there is no such thing as a CC1350 Sensortag from Texas Instruments.

The only sensortag there is is [this](http://www.ti.com/tool/tidc-cc2650stk-sensortag) one, which is based on a CC2650. So you will have to change Board to the cc26xx if I am not mistaken.

chenek, 2015-09-03:
In fact, TI has CC1350 SensorTag and I get some. I also have CC2650 SensorTag and I have no problem running Contiki on CC2650.

tonnenpinguin, 2015-09-03:
Do you have a link to that?
I cannot seem to find it unfortunately!

chenek, 2015-09-03:
It's not formal release from TI so there is no link for it.

gillesDD, 2015-09-03:
Hi,
Then, may be you coul help us about cc1350 : are you using bleeding edge
contiki 3.0 repo ?
because I strictly followed new TI wiki about cc13xx, flash xxxdemo.bin,
but still can't see any print out debug on serial consol to confirm
software has booted well
thanks in advance.

chenek, 2015-09-03:
If you mean to know how to setup Contiki running on SmartRF06EB+CC2650EM, you can refer to http://sunmaysky.blogspot.tw/2015/09/contiki-subg-hz-6lowpan-on-cc1350.html

g-oikonomou, 2015-09-03:
The supported devices are listed explicitly in the README. The one you are describing is not listed among them, ergo you are talking about an unsupported device. Closing this.

closed 2015-09-03 by g-oikonomou