contiki-os/contiki #1236: CC26xx / CC13xx - lpm sleep sequence flaws

bkozak-scanimetrics, 2015-09-03:
I've discovered a number of flaws in the way that the MCU module enters sleep mode.

These are:
1. In ```lpm_drop()``` It is possible that an ISR occurs just after calling the lpm modules' request routines which causes a peripheral (such as the radio) to turn on just before the MCU goes to sleep. The subsequent deep sleep will abruptly shut-down the peripheral.
2. In ```lpm_drop()``` it is possible that just after ```process_nevents()``` is called, an ISR fires which then polls a process (like a button ISR calling the sensor process for example). The result will be that the event will not be processed until some indeterminate point in the future (when the next ISR fires).
3. In ```lpm_drop()``` it is possible an rtc interrupt is scheduled which lies in the past (which should indicate that an etimer has expired and requires processing).
4. In ```lpm_drop()``` no rtc interrupt is scheduled if ```next_event``` is zero. As far as I can tell, zero is a valid clock time which could be the expiry time of an actual valid etimer.
5. In the radio driver in the ```transmit()``` routine, it is possible that the last command done interrupt fires just before entering into ```lpm_sleep()``` but just after checking the value of the status register. This will cause the MCU to wake up at an indeterminate point in the future or be reset due to a watchdog expiry.
6. A maximum sleep time is required for sleep mode because the watchdog can expire in sleep mode  (deep sleep mode is ok, I think, because the watchdog is shut-down).
7. Currently, ```lpm_drop``` does not attempt to schedule an rtc interrupt for the etimer if it is going into sleep mode (rather than deep sleep mode).

I realize that there may already be people working on some or all of these issues so I am opening an issue instead of creating a PR. I do, however, think that I have a pretty good idea of how to fix all of these problems and can start work on this if no-one else wants to.

g-oikonomou, 2015-09-03:
> 1.

I don't understand. Even if an interrupt does occur, the module will be subsequently notified before we drop to DS.

> 2.

Correct

> 3

I think I see what you mean

> 4

This is a problem with etimer semantics: If we don't have any scheduled etimers, `etimer_next_expiration_time()` will return 0, which - as you say - is also a valid expiration time. We need some brainstorming here, but I think getting rid of the `if(next_event)` check is probably the way forward, since my gut feeling says that we'll always have at least one etimer scheduled. This is definitely the case for IPv6 builds.

> 5.

Can you point out a condition when this could happen? I cannot see how it might.

> 6 & 7

I am somewhat out of context here. It appears to me that you are trying to sleep for long periods, whereas the logic was "sleep for short periods, deep sleep otherwise". Any particular reason why you want to sleep instead of DS? The maximum sleep time that you are mentioning is `STANDBY_MIN_DURATION`. Is that not suitable?

bkozak-scanimetrics, 2015-09-04:
> > 1.

> I don't understand. Even if an interrupt does occur, the module will be subsequently notified before we drop to DS.

You're right actually but I think there are two ways to look at this.
If we take your view then we still have a problem with the radio driver because the radio driver does not define a shutdown handler which can shut-down cleanly. Furthermore, if the lpm module is effectively able to ignore the output of ```request()``` (I know it's not quite so black and white) then what is the point in even requesting the max_pm from the modules?
If we take my view then lpm should always abide by the output of ```request()``` but then the shut-down handlers may be less necessary. I think it is better, however, to always strictly abide by the ```request()``` functions because the various drivers may have a very good reason to prevent sleep (we might be in the middle of a ContikiMAC CCA for example).

> > 5.

> Can you point out a condition when this could happen? I cannot see how it might.

The relevant code snippet:
```C
while((cmd.status & RF_CORE_RADIO_OP_MASKED_STATUS)
          == RF_CORE_RADIO_OP_MASKED_STATUS_RUNNING) {
      lpm_sleep();
    }
```
This is where I think it can occur (for example):
1. We enter the loop for the first time after sending the command, the command is still running (so we enter the body)
2. We go to sleep
3. An unrelated interrupt fires (let's say it's RTC for example); this wakes us
4. We check ```cmd.status``` (loop condition) again and find it's still not finished.
5. The command finishes and the LAST_CMD_DONE interrupt fires.
6. We go back to sleep with no incoming LAST_CMD_DONE interrupt to wake us

> > 6 & 7

> Any particular reason why you want to sleep instead of DS?

It could be that some lpm module requested ```LPM_MODE_SLEEP``` or that a user sets ```LPM_MODE_MAX_SUPPORTED``` to a lower value (perhaps for debug purposes).

> ... The maximum sleep time that you are mentioning is STANDBY_MIN_DURATION. Is that not suitable

This is not the maximum sleep time. If we get past the ```STANDBY_MIN_DURATION``` check and ```max_pm``` is ```LPM_MODE_SLEEP``` then we can go to sleep for any amount of time. I think you may have missed the second ```lpm_sleep()``` on line 282 in ```lpm_drop()```.

g-oikonomou, 2015-09-04:
> You're right actually but I think there are two ways to look at this.
I think there's only 1 way to look at it: Yours. If a module's `request_max_pm()` says "Don't Deep Sleep", then we should sleep instead. The module is then given an opportunity to shut itself down in `shutdown()`, before we power off the PD. You seem to suggest that the code does not always respect the return values of `request`. I can't see that?

One problem I _can_ think of is, perhaps the rtimer will wake up the RF after `request`, so we will end up shutting down the RFCORE PD abruptly.

> This is where I think it can occur (for example):

Understood, yes. Corner case, but still...

> This is not the maximum sleep time.

I see what you mean. There is room for improvement here.

bkozak-scanimetrics, 2015-09-04:
It looks like you accidentally put some text in quotes in your last post which shouldn't be there.

> One problem I can think of is, perhaps the rtimer will wake up the RF after request, so we will end up shutting down the RFCORE PD abruptly.

This is exactly the problem I am referring to. When I suggest that the code does not always respect the return value of ```request``` this is exactly what I mean; it is *as though* the return from ```request``` is not respected (I can see that I didn't explain this very well, sorry) because we can go into deep sleep at a point in time where some lpm module would like to prevent deep sleep via the return of ```request```.

g-oikonomou, 2015-09-04:
> It looks like you accidentally put some text in quotes in your last post which shouldn't be there.

Duh

So what we need is to prohibit a module from changing "state" (for a lack of a better word) between `request` and `shutdown`. Either that or to disable all interrupts between the two. From the entire discussion, I can see that you keen to disable interrupts earlier in `lpm_drop()`. Probably the entire thing?

bkozak-scanimetrics, 2015-09-04:
> I can see that you keen to disable interrupts earlier in lpm_drop(). Probably the entire thing?

Yes, I think that's probably the best solution here.

g-oikonomou, 2015-11-07:
I've flagged this one as a bug and I'm looking at some of these points.

The one surrounding etimer expirations is messing with my head. As you say, this check is broken:

```
 if(next_event) {
```

It can be easily replaced by

```
 if(etimer_pending()) {
```

But in that case, what to put in the `else` block? i) Bail out, ii) Deep sleep anyway without an RTC event, iii) Deep sleep and wake up in a perhaps configurable amount of ticks? I can't see what purpose iii would serve, so I'm leaning towards ii now. I thought I'd pick some brains.

g-oikonomou, 2016-01-09:
#1371 fixes some of these points.

Additionally from the discussion in the same thread:

> - if no etimer is set at all then we can sleep for at most 512 ticks.

>  Is there any reason why you can't set a wakeup at some "distant" point (say half the watchdog period) in the future if no etimer is set?

> - There is no check to make sure that we wakeup before the watchdog expires in the case where we enter lpm_sleep because a peripheral requested it, then set the wakeup based on an etimer. 

> Although, the peripheral that asked for lpm_sleep should also probably wake the MCU soon via an interrupt, this will not necessarily be possible for every peripheral. Furthermore it may not even be a peripheral which is forcing lpm_sleep, someone might have manually set ```LPM_MODE_MAX_SUPPORTED``` for debugging purposes.

g-oikonomou, 2016-07-16:
There have been multiple updates to the CC13xx/CC26xx LPM sequence recently, so I'm now closing this as outdated (hopefully most if not all fixed).

closed 2016-07-16 by g-oikonomou