contiki-os/contiki #1238: CC26xx static initial DIS interval, synchronization in larger networks

arurke, 2015-09-03:
I might be stating the obvious, but I'll post it anyways as reference/backlog - also, people might experience some issues related to it. 
I noticed I was always getting the same initial "random" DIS interval on my CC26xx sensortag each boot, this is caused by the seed being hardcoded in main(). This can cause synchronization of the DIS messages and problems in larger networks, ref 2a2175a2149e0a11af0fb5dfb514a3e5d182f46d. 

I see there is some true RNG implementation in the driverlib, but I am not confident enough on it to create a pull request atm.

Regards,
Andreas

g-oikonomou, 2015-09-03:
We need a `random.c` that implements `random_init()` and `random_rand()`. Those two would be the interface with the TRNG. Not in my immediate plans.

msnyder415, 2015-09-09:
Andreas - What about the driverlib implementation are you not confident with? 
George - Would this be architected as a cpu specific cc26xx random.c?  

I did some grepping to see how other cpu's define these functions -
./cpu/cc2430/dev/random.c:random_rand(void)
./cpu/cc2538/dev/random.c:random_rand(void)
./cpu/cc253x/dev/random.c:random_rand(void)

Using TI BLE stack as an example, I can see their hal_trng_wrapper.c shows how to use their cc26xxware TRNG API. Is this something a contiki newb like myself can help with?

g-oikonomou, 2015-09-10:
@msnyder415 Yes, this would be a `random.c` under the CPU dir. I've not had a look at TI-RTOS in a while now, but this sounds like it could be what we want (a good starting point at least). As discussed, we need those two functions plus any other `static` helpers as required, but don't try to change the `random` API.

If you are planning to work on this, to prevent back and forth, please observe naming convention and code style guidelines as per `CONTRIBUTING.md`.

closed 2016-07-16 by g-oikonomou