contiki-os/contiki #1239: How to use SmartRF06EB+CC2650EM as Slip Radio?

chenek, 2015-09-04:
I tried to build Slip-Radio.bin for SmartRF06EB+CC2650EM by "make TARGET=srf06-cc26xx BOARD=srf06/cc26xx slip-radio.bin". However, it doesn't work after I connect it to my Raspberry Pi. Does anyone know how to use  SmartRF06EB+CC2650EM as Slip Radio?

g-oikonomou, 2015-09-04:
There are some instructions in the README about how to do this: It's very straightfoward. If your Srf is not even getting detected by the RPi, then you most likely have a problem in the RPi side.

Check out the CC2538DK README for instructions how to make the XDS appear in Linux (has to do with loading the ftdi_sio kernel module): https://github.com/contiki-os/contiki/blob/master/platform/cc2538dk/README.md#for-the-smartrf06-eb-uart

chenek, 2015-09-09:
I get it work now. Thanks.

closed 2015-09-09 by chenek