contiki-os/contiki #1244: Is uip_ds6_set_addr_iid compliant with RFC 4862

vsaw, 2015-09-07:
I'm just wondering if Contikis implementation of stateless address autoconfiguration is compliant with [RFC 4862 Section 5.3](https://tools.ietf.org/html/rfc4862#section-5.3)

> A link-local address is formed by combining the well-known link-local
>   prefix FE80::0 \[[RFC4291](https://tools.ietf.org/html/rfc4291)\] (of appropriate length) with an interface
>   identifier as follows:
>
>   1.  The left-most 'prefix length' bits of the address are those of
>       the link-local prefix.
>
>   2.  The bits in the address to the right of the link-local prefix are
>       set to all zeroes.
>
>   3.  If the length of the interface identifier is N bits, the right-
>       most N bits of the address are replaced by the interface
>       identifier.

What really confuses me is the 

```C
ipaddr->u8[8] ^= 0x02;
```

where does that come from?

gillesDD, 2015-09-07:
Hi ,
I guess it is to translate from EUI64 to IID (Vendor ID) , using
universal/local (U/L) bit,
see
https://supportforums.cisco.com/document/100566/understanding-ipv6-eui-64-bit-address
Regards
gilles

adamdunkels, 2015-09-07:
Also see `Appendix A` in RFC4291:

```
   Links or Nodes with IEEE EUI-64 Identifiers

   The only change needed to transform an IEEE EUI-64 identifier to an
   interface identifier is to invert the "u" (universal/local) bit.
```

vsaw, 2015-09-07:
Awesome thanks!

closed 2015-09-07 by vsaw