contiki-os/contiki #1254: Ctimer: application callback is called even if the process is terminated.

aguedes, 2015-09-11:
Hi all,

I have been writing some Contiki applications that use the ctimer library and I've found an interesting behavior. Since I'm not sure this is an unexpected behavior, I opened this "issue" before sending any patch.

Is it expected that the application callback (provided via ctimer_set API) is called even if the application process is terminated?

In more details, is it expected that the ctimer_process switches context to the *terminated* process and calls the application callback ([link to that code](https://github.com/contiki-os/contiki/blob/master/core/sys/ctimer.c#L78-L82))? Or should we have been checking the process state (e.g. process_is_running API) before doing that?

[Here](https://github.com/aguedes/contiki/commit/fe1629b6f4d69f8a935d9dd2547da3d066e5be33) you can find a very simple application that reproduces the behavior. Even if the application process terminates (reaches PROCESS_END()), my_cb() is called. You can test it with native platform.

Thanks!

kincki, 2016-02-01:
Hi aguedes,

Can you resend the simple application code link for reproducing the case? The current link seems to be obsolete.

vsaw, 2016-02-01:
It's not a real bug that the callback is still being called it's just an unspecified corner case. The CTimer never really specified how if or how it's dependent on the processes. And to be honest I really don't know why the CTimer keeps carrying the process information with it. (especially since `ctimer_set` doesn't mention that it uses `PROCESS_CURRENT`).

Personally I would very much prefer removing the process pointer from the CTimer completely as the CTimers are working indpendantly of the process lifecycle as you pointed out.

aguedes, 2016-02-15:
Hi @kincki , sorry about that. Here follows the link https://github.com/aguedes/contiki/commit/8b9c32d3c8d48d3f261b19e652fdc818d20ee2eb .

Since this issue was opened a while ago, I re-tested and it still happens on current master.

aguedes, 2016-02-15:
Hi @vsaw, thanks for your input!

I'm not sure if removing the process information is the right way to go. According to the [Ctimer library wiki](https://github.com/contiki-os/contiki/wiki/Timers#the-ctimer-library), the callback function is expected to be executed in the same context from the process which scheduled the ctimer.

If executing the callback after the process has terminated is indeed an undesired behavior, IMO we should check process_is_running(): if true, we change the context and call the callback function. Otherwise, we remove that ctimer from 'ctimer_list' and we are done.

vsaw, 2016-02-16:
I see. In that case I think you could prepare a pull request with your proposed fix and see if the regression tests fail. If not maybe some maintainers could vote in and tell if the Pull Request if good to merge :+1:

greg-king5, 2016-02-16:
It's expected behavior, because Contiki doesn't have "automatic housekeeping".  If a process creates a persistent object, then that process (or a related process) is expected "to dispose" of the object.  Persistent objects are things such as files, allocated memory, I/O bits/pins, and timers.

Therefore, your process must stop its ctimer before that process finishes running.

```C
PROCESS_THREAD(ctimer_test_process, ev, data)
{
    /* This is needed if other processes can kill this one. */
    if(ev == PROCESS_EVENT_EXIT) {
        ctimer_stop(&timer);
    }

    PROCESS_BEGIN();

    ctimer_set(&timer, CLOCK_SECOND * 3, my_cb, NULL);

    /* other code */
    ...

    ctimer_stop(&timer);
    PROCESS_END();
}
```