contiki-os/contiki #1272: What to do with "progression"-tests? / Unsupported Capture effect

cmorty, 2015-09-23:
I'm currently looking into the capture-effect, which is currently not supported by Cooja, nor MSPSim, and probably not by Avrora either. Stronger-first can be fixed in Cooja, but stronger-last needs support by the radio-chip-model. I have a very hacky proof of concept implementation for the Sky-Mote - and a "regression" test to test my implementation. It does need some cleaning up before it is mergeable.  However I'm not really willing to put time into this, if there is no real interest.

simonduq, 2015-09-23:
I sounds like a very very welcome feature to me, but I don't hack cooja and won't be able to review/approve the code myself.

What exactly do you need from the radio chip in the stronger-last case? (simply knowing if the chip would sync to the second packet instead of ignoring it, or more than that?)

cmorty, 2015-09-23:
@simonduq: Yes, basically its telling the chip that it's now synced to a new signal. Thus adding a simple sync-lost function to the interface should probably be sufficient from Cooja's point of view (Maybe being able to query the SINR required for the Capture effect would be nice, too.).
The handling of the event is however very chip-specific, and at least the CC2420's data sheet is not very helpful (As I interpret it, it's wrong). There is also a quite sophisticated paper on this: http://dl.acm.org/citation.cfm?id=2629352
Also my current implementation brakes the Timeline, but I think that's a minor issue.
As a side note: I didn't test this, but it seems like there is also a bug in the CC2420 when not receiving at least a whole packet. This is covered up by Cooja continuing to transfer data, even if the connection is interfered.

cmorty, 2015-09-23:
Oh, and I'm currently only looking into adding a failing regression test. Raslly supporting the Capture-Effect requires some refactoring, as Cooja designed to not support the Capture effect. :(

simonduq, 2015-09-23:
OK, what does the regression test do?

cmorty, 2015-09-23:
It send two packets with a slight delay where one can be received if the capture effect is supported.

simonduq, 2015-09-23:
OK and what you want to do is add the failing test? Why?

cmorty, 2015-09-23:
As a reminder that something is wrong and as a starting-point for someone who is interested in fixing it. Kind of the Extreme Programming approach.

simonduq, 2015-09-23:
Then I'm not that much of a supporter. Capture in Cooja would be great, but as long as we don't have it I see limited gains in spending time on a regression test.

atiselsts, 2015-09-23:
The capture effect is implemented in the MRM radio model. Improving it and adding it to the other radio models would be welcome features, in my opinion.

cmorty, 2015-09-24:
@atiselsts Oh. You're talking about this implementation? https://github.com/contiki-os/contiki/blob/ef779d4fc1980a2543cd70bb96fd83f8df286369/tools/cooja/apps/mrm/java/org/contikios/mrm/MRM.java#L204-L214

cmorty, 2015-09-24:
@atiselsts And I'm pretty sure that part that is implemented breaks on sky-nodes.